robotology/yarp #1481: Methods that should be protected in RateThread

drdanz, 2017-12-04:
The following methods in `yarp::os::RateThread` are supposed to be `protected`, but are `public`instead:

```c++
virtual bool threadInit();
virtual void threadRelease();
virtual void run() = 0;
virtual void beforeStart();
virtual void afterStart(bool success);
```

I'd like to make them `protected` but this is an API change and I don't know if this will break stuff around.
@traversaro @pattacini @lornat75 @barbalberto @vtikha are you aware of any module, device or other, actually *calling* these methods? (Just *overloading* them is ok, even though I would recommend to make them protected also in derived classes)

pattacini, 2017-12-04:
@drdanz I don't believe there'll be any particular problem.
A quick test would be compiling `icub-main` with this change and see whether it'll be all fine.

drdanz, 2017-12-04:
Cool, icub-main compiles, I'll open a PR...

Nicogene, 2017-12-11:
Fixed by #1482

PR #1482 (2017-12-08): Rate thread improvements
OS/RateThread: Improvements

* Cleanup
* Avoid useless call
* Improve PIMPL implementation
* Remove useless private method

OS/RateThread: Protect methods that should not be public.

The following methods in `yarp::os::RateThread` are now `protected` instead
of `public`:
* `virtual bool threadInit();`
* `virtual void threadRelease();`
* `virtual void run() = 0;`
* `virtual void beforeStart();`
* `virtual void afterStart(bool success);`

Fixes #1481
commit a84e2a7b (2017-12-08): OS/RateThread: Protect methods that should not be public.

The following methods in `yarp::os::RateThread` are now `protected` instead
of `public`:
* `virtual bool threadInit();`
* `virtual void threadRelease();`
* `virtual void run() = 0;`
* `virtual void beforeStart();`
* `virtual void afterStart(bool success);`

Fixes #1481
closed 2017-12-11 by Nicogene