robotology/yarp #1496: YARP devel with YARP_USE_OPENCV set to OFF does not compile

traversaro, 2017-12-18:
With the following error: 
~~~
[ 91%] Built target YARP_OS
[ 91%] Built target YARP_init
[100%] Built target YARP_sig
Scanning dependencies of target yarpdatadumper
[100%] Building CXX object src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/main.cpp.o
/Users/traversaro/src/something-superbuild/isaac/YARP/src/yarpdatadumper/main.cpp:305:9: error: member initializer 't0' does not name a non-static data member or base class
        t0(0.0)
        ^~~~~~~
1 error generated.
make[3]: *** [src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/main.cpp.o] Error 1
make[2]: *** [src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/all] Error 2
make[1]: *** [src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/rule] Error 2
make: *** [yarpdatadumper] Error 2
~~~

traversaro, 2017-12-18:
I guess this is related to this commit https://github.com/robotology/yarp/pull/1495/commits/2ceaac3d5effa9a6bc5ac6488055a964369aba88 in this PR https://github.com/robotology/yarp/pull/1495 . @damn1

Nicogene, 2017-12-20:
Fixed by #1497

PR #1497 (2017-12-19): fix bug introduced by initialization in yarpdatadumper
Fixes #1496
closed 2017-12-20 by Nicogene