robotology/yarp #1502: Cannot view IVelocityControl on fakeMotionControl device

jgvictores, 2017-12-21:
Identified by @PeterBowman [here](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/54#issuecomment-353431240). Cannot view `IVelocityControl` on `fakeMotionControl` device. However, *can* view `IVelocityControl2`.

This is strange, given [IVelocityControl2 inherits publicly from IVelocityControl](https://github.com/robotology/yarp/blob/8be516188ec354f82e4e6852a61364a16ef288d7/src/libYARP_dev/include/yarp/dev/IVelocityControl2.h#L31)...
Maybe something happening [here](https://github.com/robotology/yarp/blob/8be516188ec354f82e4e6852a61364a16ef288d7/src/devices/fakeMotionControl/fakeMotionControl.cpp#L356) but see nothing strange [here](https://github.com/robotology/yarp/blob/8be516188ec354f82e4e6852a61364a16ef288d7/src/libYARP_dev/include/yarp/dev/ControlBoardInterfacesImpl.inl#L214-L336)....

Small Python test snippet:
```python
import yarp
yarp.Network.init()
options = yarp.Property()
options.put('device','fakeMotionControl')
dd = yarp.PolyDriver(options)
dd.isValid()
pc = dd.viewIPositionControl()
print pc  # <yarp.IPositionControl; proxy of <Swig Object of type 'yarp::dev::IPositionControl *' at ....> >
vc = dd.viewIVelocityControl()
print vc  # None
```

PeterBowman, 2017-12-21:
I just commented out the inheritance from `ImplementVelocityControl<FakeMotionControl, IVelocityControl>` and related constructor/uninitialize()/uninitialize() calls, works fine when viewing on `IVelocityControl`.

PeterBowman, 2017-12-21:
In my attempt to understand what is going on, I mimicked the inheritance chain of `IVelocityControl` and `IVelocityControl2` in the following snippet:

```c++
#include <iostream>

// IVelocityControl
class Base
{
    public:
        virtual void foo() = 0;
};

// IVelocityControl2
class Base2 : public Base
{
    public:
        virtual void bar() = 0;
};

// ImplementVelocityControl
class ImplA : public Base
{
    public:
        virtual void foo() {}
};

// ImplementVelocityControl2
class ImplB : public Base2
{
    public:
        virtual void foo() {}
        virtual void bar() {}
};

// DeviceDriver
class Driver
{
    public:
        virtual void baz() {}
};

// FakeMotionControl
class Test : public Driver, public ImplA, public ImplB {};

int main(int argc, char ** argv)
{
    Driver *d = new Test;  // internals of PolyDriver::open
    Base *b = dynamic_cast<Base *>(d);  // DeviceDriver::view

    if (b == 0)
        std::cout << "null" << std::endl;

    delete d;
    return 0;
}
```

Like in our case, `dynamic_cast` fails and `null` is printed. However, if the first assignment is changed to:

```c++
Test *d = new Test;
```

then we get a compiler error on the next line: `‘Base’ is an ambiguous base of ‘Test’`. This is, most likely, the [diamond problem](https://isocpp.org/wiki/faq/multiple-inheritance#mi-diamond) caused by multiple inheritance with a common base class.

I'd postulate that deriving from `ImplementVelocityControl` in `FakeMotionControl` is not just superfluous, but also harmful: it hides unwanted behavior that is not visible on compile time, only unveiled by RTTI if the proper checks are undertaken.

drdanz, 2018-01-09:
Perhaps we should be using virtual inheritance here?

PeterBowman, 2018-01-09:
I guess you mean:

```c++
class IVelocityControl2 : public virtual IVelocityControl { /* ... */ };

class ImplementVelocityControl : public virtual IVelocityControl { /* ... */ };

class MyClass : public ImplementVelocityControl2,
                public ImplementVelocityControl // more "safe" now?
{};
```

I did that with the `IVelocityControl2` and `ImplementVelocityControl` counterparts in the example above. The compiler threw this: `error: no unique final overrider for ‘virtual void Base::foo()’ in ‘Test’`. See also: [SO](https://stackoverflow.com/q/616380).

PR #1504 (2018-01-26): Avoid multiple implementation of the IVelocityControl interface
Resolves #1502. By looking at the sources, `ImplementVelocityControl<>` seems clearly redundant (and causes problems as stated in the linked issue) when deriving from `ImplementVelocityControl2` at the same time.

P.S. No changelog file available in `doc/release/`.
closed 2018-01-26 by barbalberto