robotology/yarp #1509: Move Companion out of YARP_OS

drdanz, 2018-01-09:
I suggest we make a separate library for that.

drdanz, 2018-01-09:
Also the yarprun related classes should be moved in that library

Nicogene, 2018-06-01:
Closed by #1711

closed 2018-06-01 by Nicogene

PeterBowman, 2018-06-02:
It looks like Eigen is a hard dependency for the (mandatory) YARP_companion library:

https://github.com/robotology/yarp/blob/c18e29b4253c76883fac15986449630487588cc4/src/libYARP_companion/CMakeLists.txt#L31

...which conflicts with the following recent statement (https://github.com/robotology/yarp/issues/1559#issuecomment-391650201):

> As discussed, YARP_math will still be optional, but, if eigen is found, the default value will be ON

In fact, I'm unable to compile latest devel (https://github.com/robotology/yarp/commit/c18e29b4253c76883fac15986449630487588cc4) on a system with no Eigen3 installed, see [this Travis build](https://travis-ci.org/roboticslab-uc3m/tools/jobs/387127528#L811). Anyway, those `include_directories()` calls are probably superfluous (and misused given the modern `target_include_directories()` alternative) since grepping `ace` and `eigen` doesn't produce any relevant result in this library's source tree.

drdanz, 2018-06-02:
I think the include_directories was fixed in #1718

drdanz, 2018-06-03:
@PeterBowman The problem was that you can still install YARP_eigen if eigen is not available, but you will still need eigen3 in order to use it. Now it should be fixed, but if you want to install YARP and install eigen3 later you will need eigen3 3.3.1 or later, i.e. this won't work on ubuntu trusty. I don't think this is going to be a real problem...

PeterBowman, 2018-06-03:
That's https://github.com/robotology/yarp/commit/1f0fcb36104d16ca6d7d364d3e9734c00b86a34e. Thanks! (and sorry for the OT)

drdanz, 2018-06-03:
No problem :wink:

PR #1711 (2018-05-31): move Companion out of YARP_OS
Fixes #1509 

In order to remove any `libYARP_OS` dependences from the `Companion` some changes were necessary:
* Some `Companion` functions has been moved to `NetworkBase`
    * `disconnectInput`
    * `poll`
    * `sendMessage`
    * `wait` (the `wait()` has been splitted into two new functions `waitConnection()` and `waitPort()` to keep code clean and readable.

* `Companion::exists()` has been removed unifying its code with `NetworkBase::exists()`. A parameter has been added to manage the last check in the function. 

* `readString()` function has been depreca [... cut]