robotology/yarp #1514: Setting yarpserver ip and port within code

dcam0050, 2018-01-16:
More of a request for help rather than an issue.

I have compiled Yarp in C# to integrate it with the Unity Game Engine and it works well however I would like to deploy the game without requiring the user install yarp on their device.

The device will however be present on a local network that has a yarpserver running on another computer. My question is how do you set the ip address and port for yarpserver in the code such that it can be found during a Network.init()?

Nicogene, 2018-01-16:
Hi @dcam0050 !
check this : http://yarp.it/classyarp_1_1os_1_1NetworkBase.html#a68a57b1493c8d9ce73b22801a6c32f85 .
I never used it but it could it what you are looking for.
Let me know if it solves your problem!

francesco-romano, 2018-01-17:
@dcam0050 you can also take a look [here](https://github.com/francesco-romano/iYarp/blob/master/libs/yarp_iOS/yarp_iOS/Utilities/IITYarpNetworkConfiguration.mm#L67)
even if I wrote that code almost 3 years ago

dcam0050, 2018-01-18:
@Nicogene The setNameServerContact method works perfectly. Thanks!

closed 2018-01-18 by dcam0050