robotology/yarp #1528: crash on harness tests when compiled static and yarp::os::Network object is created inside

aerydna, 2018-01-31:
By compiling YARP libraries statically and putting a `yarp::os::Network` object in a `UnitTest` class as a member, the `harness_[lib]` application will crash. follows instructions:

1. in cmake, enable the tests
2. in cmake set `BUILD_SHARED_LIBS` to `ON`
3. modify any of the yarp's UnitTest classes putting as a member a `yarp::os::Network` object (example the `BottleTest` class inside `tests/libYARP_OS/BottleTest.cpp`)
4. build
5. launch the test (example `harness_os regression`)
6. enjoy the `segfault`

traversaro, 2018-01-31:
> in cmake set BUILD_SHARED_LIBS to ON 

ON or OFF ?

aerydna, 2018-01-31:
@traversaro sorry.. it was OFF

drdanz, 2018-02-09:
Just for curiosity, why would you want to add a Network object as a member in a test? the network object is already created in the main...

aerydna, 2018-02-12:
i had it in my personal test and i forgot to delete it.. nothing more.  but since the network object is created in some rtf plugin i thougt it was worth to report it. Ignore and close if inappropriate