robotology/yarp #1531: Using YARP devel from a C++14 codebase result in warning related to YARP_DEPRECATED

traversaro, 2018-02-05:
Using YARP in a compilation unit that is compiled by gcc with `-std=gnu++14` in Ubuntu 16.04 is resulting in the following warnings:
~~~
/home/sdafarra/Software/robotology-superbuild/build/install/include/yarp/os/PortReaderBufferBase.h:74: an attribute that appertains to a type-specifier is ignored
     void YARP_DEPRECATED setAllowReuse(bool flag = true);
          ^
~~~

traversaro, 2018-02-05:
cc @S-Dafarra

drdanz, 2018-02-09:
I think (but I might be wrong) that the `__attribute__(deprecated)` and the `[[deprecated]]` options have to be set in different places (i.e. before or after the return type). Further investigation is required.

traversaro, 2018-03-14:
Apparently this is creating an error in the latest version of macOs / homebrew : https://github.com/robotology/gazebo-yarp-plugins/issues/351 .

S-Dafarra, 2018-03-16:
Using the ``devel`` branch now it fails also in Linux with the following error:
```
In file included from /usr/local/src/robot/yarp/build/src/libYARP_conf/include/yarp/conf/api.h:13:0,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/api.h:13,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Runnable.h:13,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/RateThread.h:13,
                 from /usr/local/src/robot/element_mpc-for-walking/code/icubWalkingV2/include/WalkingReferenceInterfaceThread.h:4,
                 from /usr/local/src/robot/element_mpc-for-walking/code/icubWalkingV2/include/WalkingPatternGeneratorUnicycle.h:26,
                 from /usr/local/src/robot/element_mpc-for-walking/code/icubWalkingV2/src/WalkingPatternGeneratorUnicycle.cpp:14:
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/PortReaderBufferBase.h:77:10: note: an attribute that appertains to a type-specifier is ignored
     void YARP_DEPRECATED setAllowReuse(bool flag = true);
          ^
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/PortReaderBufferBase.h:78:10: note: an attribute that appertains to a type-specifier is ignored
     void YARP_DEPRECATED release(yarp::os::PortReader* completed);
          ^
In file included from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/api.h:13:0,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Runnable.h:13,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/RateThread.h:13,
                 from /usr/local/src/robot/element_mpc-for-walking/code/icubWalkingV2/include/WalkingReferenceInterfaceThread.h:4,
                 from /usr/local/src/robot/element_mpc-for-walking/code/icubWalkingV2/include/WalkingPatternGeneratorUnicycle.h:26,
                 from /usr/local/src/robot/element_mpc-for-walking/code/icubWalkingV2/src/WalkingPatternGeneratorUnicycle.cpp:14:
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Stamp.h:96:7: error: expected identifier before ‘__attribute__’
 class YARP_OS_DEPRECATED_API Stamped
       ^
In file included from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/all.h:56:0,
                 from /usr/local/src/robot/element_mpc-for-walking/code/icubWalkingV2/src/WalkingPatternGeneratorUnicycle.cpp:15:
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Stamp.h:98:1: error: expected primary-expression before ‘public’
 public:
 ^
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Stamp.h:98:1: error: expected ‘}’ before ‘public’
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Stamp.h:98:1: error: expected ‘,’ or ‘;’ before ‘public’
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Stamp.h:108:30: error: ‘virtual’ outside class declaration
     virtual Stamp getStamp() const = 0;
                              ^
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Stamp.h:108:30: error: non-member function ‘yarp::os::Stamp yarp::os::getStamp()’ cannot have cv-qualifier
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Stamp.h:108:38: error: function ‘yarp::os::Stamp yarp::os::getStamp()’ is initialized like a variable
     virtual Stamp getStamp() const = 0;
                                      ^
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Stamp.h:112:1: error: expected declaration before ‘}’ token
 } // namespace yarp
 ^
CMakeFiles/icubWalkingPatternGeneratorUnicycle.dir/build.make:62: recipe for target 'CMakeFiles/icubWalkingPatternGeneratorUnicycle.dir/src/WalkingPatternGeneratorUnicycle.cpp.o' failed
make[2]: *** [CMakeFiles/icubWalkingPatternGeneratorUnicycle.dir/src/WalkingPatternGeneratorUnicycle.cpp.o] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/icubWalkingPatternGeneratorUnicycle.dir/all' failed
make[1]: *** [CMakeFiles/icubWalkingPatternGeneratorUnicycle.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

traversaro, 2018-03-16:
https://github.com/robotology/yarp/pull/1594 solves the issue?

S-Dafarra, 2018-03-16:
No, I had to remove manually ``YARP_OS_DEPRECATED_API`` from line https://github.com/robotology/yarp/blob/devel/src/libYARP_OS/include/yarp/os/Stamp.h#L96.

traversaro, 2018-03-16:
I think there are two different problems, one is the `YARP_DEPRECATED` one (solved by  https://github.com/robotology/yarp/pull/1594 , if I understood correctly) and the `class YARP_OS_DEPRECATED_API Stamped` is a different one.

closed 2018-03-22 by drdanz