robotology/yarp #1538: GazeboYarpPlugins::ClockServer After reset time problems

triccyx, 2018-02-09:
After doing GazeboYarpPlugins::ClockServer::reset the simulation time in Gazebo is correctly reset together with all the simulation paramiters; while yarp::os::NetworkClock::now() time is not.

This creates a misalignment between the two entity and creates problems when a yarp::os::NetworkClock::delay() is called.

Nicogene, 2018-02-09:
I think that it is a @barbalberto job 😉 😬

barbalberto, 2018-02-12:
`NetworkClock` was not designed to handle clock reset, so yes, probably it will not work.
`NetworkClock::now()` does not need to be reset, because it always returns the last clock msg received from the server, therefore if the server goes back in time, `::now()` will follow. Only reason for the `now()` not to be updated is that the server does not broadcast time after the reset.

In the current implementation, if a reset is called when a thread is already sleeping in a `delay`, that thread will not wake up properly.