robotology/yarp #1552: Unknown CMake command "swig_add_library" error, while compiling bindings

jgvictores, 2018-02-17:
Related: #1178

The main issue here is that from [here](https://cmake.org/pipermail/cmake-developers/2017-April/029946.html) we see `swig_add_library` was introduced in CMake 3.8.0, while [http://wiki.icub.org/wiki/YARP_Supported_Distributions](http://wiki.icub.org/wiki/YARP_Supported_Distributions) indicates it's safe to use CMake 3.0.2 (maybe say 3.2.2, or at least Xenial's 3.5.1).

Current `master` (YARP 2.3.72) seems to use `swig_add_library` consistently in a mandatory fashion (no conditionals), at least for [python](https://github.com/robotology/yarp/blob/935eddd4f12c0fd766bdd7e98440e6a395ad322d/bindings/python/CMakeLists.txt#L33) and [java](https://github.com/robotology/yarp/blob/935eddd4f12c0fd766bdd7e98440e6a395ad322d/bindings/java/CMakeLists.txt#L56).

jgvictores, 2018-02-17:
More refs: https://cmake.org/cmake/help/v3.8/module/UseSWIG.html (and switching to CMake version 3.7  and below).

jgvictores, 2018-02-17:
...goes back to commit 86b118c91fa3c84b22a43597db5c2ebde4dd362f : `Bindings: Use swig_add_library instead of swig_add_module; This is now in YCM UseSWIG version`.

Will try to trace back which UseSWIG version is being used...

jgvictores, 2018-02-17:
There was actually a pretty live discussion at mentioned commit 86b118c, but exactly the issue in this case.

Solution on my machines (CMake 3.5.1): the issue was, in fact, having a previously YCM version installed (for accelerating ycm-dependent builds, even if updated to `master`). Some strange shadowing must have been happening such that the `/usr/share/cmake/Modules/UseSWIG.cmake` was being invoked rather than any YCM version.

@drdanz et Al: Feel free to close this issue for the historic, and/or open a new [YCM issue](https://github.com/robotology/ycm/issues) if you think this is relevant.

jgvictores, 2018-04-06:
The issue the first case, compiling the bindings from yarp build, and still persists. It's easy to replicate:
1. Install YCM (https://github.com/robotology/ycm)
2. Download fresh YARP (https://github.com/robotology/yarp)
3. During cmake step of `mkdir build && cd build && cmake .. -DYARP_COMPILE_BINDINGS=ON -DCREATE_PYTHON=ON` output:
```
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable exact version "2.7.12") 
CMake Error at bindings/python/CMakeLists.txt:33 (swig_add_library):
  Unknown CMake command "swig_add_library".
-- Configuring incomplete, errors occurred!
```

Obscure shadowing, right?

PS: The external build fix will probably be good for the workflow I follow on Windows (broken as of YARP 2.3.72 (2018-02-12), which involves downloading binaries and compiling bindings separately. It's a nice workflow against which I would not put error messages, we can get back to that in the future.

PeterBowman, 2018-06-05:
This is an upstream issue solved at current YCM master, see https://github.com/robotology/ycm/pull/139.

Nicogene, 2018-06-06:
> This is an upstream issue solved at current YCM master, see robotology/ycm#139.

Shall we close this issue?

PeterBowman, 2018-06-06:
Sure. Let me give more visibility to https://github.com/robotology/ycm/pull/146, which is a follow-up for future YCM 0.8.

jgvictores, 2018-06-07:
I can confirm that the steps described [above](https://github.com/robotology/yarp/issues/1552#issuecomment-379316266) no longer lead to the described error: both configure and compilation are successful (same distro: Ubuntu Xenial).

closed 2018-06-07 by Nicogene