robotology/yarp #1556: PWM and Current data position seem to be swapped in the yarp port StateExt

nunoguedelha, 2018-02-22:
After doing some tests on the friction estimation and potting joint torques against PWM, we noticed that the PWM values where too small (<20), while we were expecting something closer to hundreds. Looking at the raw data, we realized that the PWM and Current fields are swapped, as we can see in the data dumper sample below:
```
# frame n. and timestamp
15498 1518207399.670650

# joint q, dq, ddq
(-0.0494385972622003 9.953637582123 0.0384522423150447 -0.0659181296829338 0.236206631363846 -0.0823976621036672) [ok] 
(-0.0549317747357782 0.0 -0.0549317747357782 0.0 0.0 0.0549317747357782) [ok] 
(0.0 0.0 0.0 0.0 0.0 0.0) [ok] 
# motor q, dq, ddq
(2240.12876007998 330.925490540748 -138.005097668695 -2957.55971083914 1475.47845575795 1218.67790204566) [ok] 
(5.49317747357782 0.0 5.49317747357782 0.0 0.0 5.49317747357782) [ok] 
(0.0 0.0 0.0 0.0 0.0 0.0) [ok] 
# Joint Torques, PWM and Current
(-0.303905 2.829681 -0.113787 -0.927275 -0.366501 0.17449) [ok] ---> torques
(1.33125 1.9875 1.084375 -0.515625 -1.025 0.1125) [ok]  ------------> PWM (expected)
(632.0 1115.0 331.0 70.0 -94.0 7.0) [ok]   -------------------------> Current (expected)

(7565168 7565168 7565168 7565168 7565168 7565168) [ok] (1718187123 1718187123 1718187123 1718187123 1718187123 1718187123) [ok]
```

But clearly, `632.0 1115.0 331.0 70.0 -94.0 7.0` are PWM values and not current. Same result seen in a data dumper log today.

nunoguedelha, 2018-02-22:
@marcoaccame @valegagge can you take a look at this?

CC @traversaro

drdanz, 2018-02-22:
CC @barbalberto

traversaro, 2018-02-22:
I think the problem is in the icub-main software (embObjLib or underneath) .

valegagge, 2018-02-23:
Hi @nunoguedelha ,
I'd like to remeber you that the value of PWM are in %. see 
http://www.yarp.it/classyarp_1_1dev_1_1IPWMControl.html#a02408ad9827f355f461346ec5859f93d,

while the currents are in milliAmpere.

traversaro, 2018-02-23:
I see, the semantics of the measurement changed during the migration from PIDOutput to PWM interfaces (https://github.com/robotology/yarp/pull/1145). 

The problem is now that the estimation software will estimate the PWM (in percentage) ---> Torque (Nm) coefficient, but the interface for setting torque control parameters and the embObj configuration files wants the Torque (Nm) ---> PWM ( as raw integer ) coefficient (or something similar, see https://github.com/robotology/robots-configuration/blob/master/iCubGenova04/hardware/motorControl/right_leg-eb9-j4_5-mc.xml#L86 ). However, this definitely a icub-main issue and only partially a YARP issue (the only part of this issue related to YARP is the precise semantics of the `ktau` parameter http://www.yarp.it/classyarp_1_1dev_1_1MotorTorqueParameters.html#a7099e7a5441c7c555dd4c936fae23e46 .

valegagge, 2018-02-23:
@traversaro you are right!

We need to update API (or use new interface) in order to use PWM in %.

@nunoguedelha 
You can use the full scale in the configuration files to convert from raw to %. 
Use parameter` fullscalePWM`; see https://github.com/robotology/robots-configuration/blob/8423a4d9ef054ed579b67dccac1cc34ec9105e5c/iCubTemplates/iCubTemplateV4_0/hardware/mechanicals/body_part-ebX-jA_B-mec.xml#L12 .

As a temporary solution, I can add a remote variable for fullscalePWM, but I can't implement it immediately

traversaro, 2018-02-23:
@valegagge Closing the issue here as it is not YARP-related, I already copied your reply in https://github.com/robotology/icub-main/issues/497 .

closed 2018-02-23 by traversaro

nunoguedelha, 2018-02-23:
@valegagge @traversaro Thanks!

randaz81, 2018-05-03:
fixed by https://github.com/robotology/yarp/pull/1612