robotology/yarp #1559: Make YARP_math mandatory

drdanz, 2018-02-23:
Once upon a time, YARP_math depended on GSL that is GPL, therefore building YARP_math was optional.

This dependency is now replaced by Eigen and we no longer have issues with the licenses, therefore the only reason for keeping the math library optional is that we don't want to add mandatory dependencies on YARP

Having YARP_math optional has some weird effects. For example in YARP_dev some parts are optional due to this limitation, yarpmanager and some other parts of YARP behave in a different way if the library is not built.

All our instructions, suggest to enable the `CREATE_LIB_MATH` flag when building YARP and we actually do it in all our computers.

I suggest we should stop pretending that we depend only on ACE, and make Eigen a mandatory dependency for building YARP. Eventually it is very simple to include it in the extern folder, since it is a header only library.

Comments?
@lornat75 @traversaro @pattacini @vtikha @randaz81

barbalberto, 2018-02-23:
Please remember that we must compile yarp in such devices like R1 head which is a ARMhf with very limited capabilities and very slow compilation. (like 1h to compile yarp & icub, I'm not kidding)
That device does not use yarp math in any way, so having it optional will same a lot of complation time.

I'd propose to have yarp math ON by default, but it'll be nice to disable it where it is just a pain o compile.

Nicogene, 2018-02-23:
I agree with @barbalberto , I had an experience on compiling YARP on poor hardware resources so I'd prefer to put it as an option set to ON by default.

traversaro, 2018-02-23:
+1 for documenting Eigen as a dependency as much as possible, but still support disabling it (as we do for ACE anyway).

drdanz, 2018-02-23:
In this case, I'd like to propose another solution... split dev in 2 libraries, one depending on math, and the other not depending on it.

lornat75, 2018-02-23:
Which part of dev depends on libYARP_math? I hope it is just plugins. If it is part of the dev library we should make math mandatory.

barbalberto, 2018-02-23:
yarp math is used by `imu_bosh` plugin and the following wrappers
```
if (CREATE_LIB_MATH)
  list(APPEND YARP_devices_SRCS src/devices/FrameTransformClient/FrameTransformClient.cpp
                                src/devices/FrameTransformServer/FrameTransformServer.cpp
                                src/devices/Localization2DClient/Localization2DClient.cpp
                                src/devices/Map2DClient/Map2DClient.cpp
                                src/devices/Map2DServer/Map2DServer.cpp
                                src/devices/Navigation2DClient/Navigation2DClient.cpp
                                src/devices/Rangefinder2DClient/Rangefinder2DClient.cpp)
endif()
```

My idea was simply to same compilation time but it's not a big issue. 
If making the library mandatory simplifies the maintenance of the repo, I vote for the mandatory instead.

drdanz, 2018-02-23:
Unfortunately, it is used also by the relative interfaces, some methods return objects from YARP_math

```cmake
if (CREATE_LIB_MATH)
  list(APPEND YARP_dev_HDRS include/yarp/dev/IFrameTransform.h
                            include/yarp/dev/ILocalization2D.h
                            include/yarp/dev/IMap2D.h
                            include/yarp/dev/INavigation2D.h
                            include/yarp/dev/Map2DLocation.h
                            include/yarp/dev/MapGrid2D.h)
endif()
```

This means that depending on which interfaces are used, when using YARP_dev, YARP_math should or should not be linked

drdanz, 2018-02-23:
Also MapGrid2D is a data type, not an interface, it's right place is either in sig or maybe math, not in dev

lornat75, 2018-02-23:
👎 libYARP_dev should only use objects from libYARP_sig.  how about:
1) moving objects returned by these interfaces to libYARP_dev
or
2) making libYARP_math mandatory and part of the core libraries
or
3) splitting libYARP_dev in two libraries

depending on the feasibility of option 1) it would be my preference, otherwise 2) then 3).

drdanz, 2018-02-23:
## Direct Dependencies:

* `IFrameTransform.h` includes `<yarp/math/Quaternion.h>`
* `Map2DLocation.h` includes `<yarp/math/Vec2D.h>`
* `MapGrid2D.h` includes `<yarp/math/Vec2D.h>`

## Indirect Dependencies
* `ILocalization2D.h` includes `<yarp/dev/Map2DLocation.h>`
* `INavigation2D.h` includes `<yarp/dev/Map2DLocation.h>`
* `IMap2D.h` includes `<yarp/dev/Map2DLocation.h>` and `<yarp/dev/MapGrid2D.h>`

Nicogene, 2018-02-23:
Another solution could be to move that classes in another library like `libYARP_service` or `libYARP_randaz` :stuck_out_tongue: 
All joking aside, these classes are not strictly related to physical devices or am I wrong ?

drdanz, 2018-02-23:
That's one of the reason why I was suggesting to split stuff related to physical devices and virtual devices

Also I forgot to mention that, in order to avoid dependencies on math in several devices still in libYARP_dev, quaternion operations and some other mathematical operations are defined in `yarpRosHelper.h`

barbalberto, 2018-02-23:
I see what you mean, now.
Maybe we could move data types in sig as @lornat75  suggested and move devices which are not simply 'interface wrappers' to a different lib. For example mapServer and Transform server are real servers performing a job on its own (they _do not_ have subdevice), so they looks more like tools or modules (who can still be instantiated as plugins).

lornat75, 2018-03-23:
Going back to the original question, I think we should keep `math' optional but clean the dependencies so that libYARP_sig does not depend on libYARP_math.

drdanz, 2018-05-24:
As discussed, YARP_math will still be optional, but, if eigen is found, the default value will be ON

closed 2018-05-24 by drdanz