robotology/yarp #1571: Resetting a YARP controlboad to a different joint configuration

lothas, 2018-03-02:
I wrote a gazebo plugin a while ago to reset the simulation and reset my model to a desired position and/or joint configuration. Up until now it seemed that everything worked OK because I was always resetting to the initial configuration stated in the sdf model file.
I'm now trying to reset my model to a different desired joint configuration. My gazebo plugin correctly updates the model but the gazebo-yarp-plugin always resets to the value specified in the sdf file, quickly driving the joint positions to those ones.
Is there a way of programmatically specifying the desired "OnReset" configuration for a yarp control board?

jgvictores, 2018-03-05:
Maybe [yarp::dev::IEncoders::setEncoders](http://www.yarp.it/classyarp_1_1dev_1_1IEncoders.html#a8c5adefdc8c0627a7ecfe7f6d972deaa) is what you need?

traversaro, 2018-03-05:
Hi @lothas , I am afraid the "reset to arbitrary configuration" functionality is not supported in gazebo-yarp-plugins (moving the issue there as it is a gazebo-related problem).

traversaro, 2018-03-05:
Issue moved to [robotology/gazebo-yarp-plugins #349](https://github.com/robotology/gazebo-yarp-plugins/issues/349) via [**ZenHub**](https://www.zenhub.com/)

closed 2018-03-05 by traversaro