robotology/yarp #1573: YARP master is not working with CMake 3.2.2

traversaro, 2018-03-03:
By chance, I tested the current YARP master with CMake 3.2.2 and I got the following error:
~~~
$ cmake -G"${TRAVIS_CMAKE_GENERATOR}" -DCMAKE_BUILD_TYPE=${TRAVIS_BUILD_TYPE} ..
CMake Error at cmake/ycm-0.6.0/modules/GitInfo.cmake:379 (cmake_parse_arguments):
  Unknown CMake command "cmake_parse_arguments".
Call Stack (most recent call first):
  cmake/YarpVersion.cmake:34 (git_wt_info)
  CMakeLists.txt:38 (include)
~~~
https://travis-ci.org/robotology/idyntree/jobs/348595108#L1518

As we are not currently testing this old CMake version anywhere, I think it could make sense to just bump the minimum required cmake version in the CMakeLists.txt of the project for getting at least a clearer error message.

drdanz, 2018-03-06:
We should fix this on master, though...
On devel it is safe to bump the dependency to 3.5

closed 2018-03-30 by drdanz

drdanz, 2018-03-30:
I fixed this only in master, devel is probably not building right now with CMake < 3.5, but that's the plan, therefore I'll update the `cmake_minimum_required` as soon as I can.

commit 89402096 (2018-03-30): CMake: Import support CMakeParseArguments and FindPackageHandleStandardArgs from YCM

Fixes #1573