robotology/yarp #1575: Using Catch2 instead of our custom unit test system

drdanz, 2018-03-06:
I made some tests porting YARP unit tests to Catch2, see https://github.com/drdanz/yarp_test_catch2

Porting seems to be really quick and easy, and the code is cleaner. The library is just one header file.
I think it is worth porting.

What do you think?

traversaro, 2018-03-06:
If it is doable without much effort, I would try to use catch **1** to avoid embedding a copy of the library (~500 kb) in YARP, as we could (at least in Ubuntu) use the version provided by apt : 
https://repology.org/metapackage/catch/versions .

drdanz, 2018-03-06:
Porting "as is" is just enough to get everything working, I don't think a full refactor is needed.
Of course adding more tests is always welcome :smile:

Also setting up a coverage infrastructure would be really useful

lornat75, 2018-03-06:
I like "porting as is is just enough" 👍

drdanz, 2018-03-06:
About catch1, I made a quick test, everything seems compatible with catch from Debian, except for [adding command line options](https://github.com/catchorg/Catch2/blob/master/docs/own-main.md#adding-your-own-command-line-options) (I used it in order to support the --yarp-verbose argument (verbose was already taken by catch), see https://github.com/drdanz/yarp_test_catch2/commit/4b49f692ae23ff951c60854766d8eaa789d4983b

Also a FindCatch.cmake is required, since there is not one.

@lornat75 If you really want some refactor, one improvement could be not using "impl" in unit tests, or separate public libraries and impl unit tests. In this way we could actually filter out impl symbols from the ABI

traversaro, 2018-03-06:
Not blocking this activity in any way, but it may be worth mentioning that catch is not currently supporting parametrized tests, see https://github.com/catchorg/Catch2/issues/850 .

drdanz, 2018-03-06:
That's one limit in the Catch library, but I'll add that `yarp::os::impl::UnitTest` does not support that as well...

Nicogene, 2018-03-06:
Probably a silly question, Catch seems a nice testing framework, but why we do not migrate to **our** testing framework??

drdanz, 2018-03-06:
RTF is not an "unit testing framework", it is an "integration test framework" (I just made up the two definitions)

Nicogene, 2018-03-06:
:thinking: I see, probably RTF is an overkill for what we need

drdanz, 2018-03-06:
@traversaro catch1 from system is now supported in master :see_no_evil: 

I added a check on `CATCH_VERSION_MAJOR` that actually exists since Catch v2.1.2, therefore that part is left out even on Catch2 2.1.1 and previous releases even though they would support it. Anyway, not a real issue.

traversaro, 2018-03-21:
An additional reason for this: I recently added a new test for libYARP_dev, and the test was not added to CTest until I removed the build directory and started a fresh build.

drdanz, 2019-03-22:
THis was fixed long time ago. Closing

closed 2019-03-22 by drdanz