robotology/yarp #1579: Set YARP_WRAP_STL_STRING default value to OFF also on MSVC platform

traversaro, 2018-03-08:
Having two different default values of `YARP_WRAP_STL_STRING` for Microsoft Visual Studio and the other platforms, as introduced in https://github.com/robotology/yarp/pull/1406/commits/5595189d93bb331eeef595a11d721ab22460b036, means that we are effectively shipping two different API, meaning that is quite easy to write code that works for an API, but does not work for another API. 

This is causing several problems: 
* https://github.com/robotology-playground/yarp-matlab-bindings/issues/14
* https://github.com/robotology-playground/yarp-matlab-bindings/issues/17
* https://github.com/robotology/wb-toolbox/issues/98

 This is already planned as discussed in https://github.com/robotology/yarp/pull/1406#issuecomment-336433473 , this is just a issue to avoid forgetting about it.

drdanz, 2018-03-09:
This is an easy task, but we should take care to get rid of msvc warnings

traversaro, 2018-03-09:
Note that, as long

traversaro, 2018-03-09:
Related to suppressing the C4251 warning suppression, check this related discussion: 
* https://bitbucket.org/ignitionrobotics/ign-rndf/pull-requests/6/ign-cmake/diff#comment-53396887 
* https://bitbucket.org/ignitionrobotics/ign-common/pull-requests/69/refactoring-the-cmake-scripts-to-use/diff#comment-49339331
I think this is already the case, but we should properly document that the YARP C++ library is only verified to work if both the downstream code and YARP are compiled with the same compiler (i.e. Visual Studio or Mingw, you cannot mix them, unless this is explicitly supported like with Visual Studio 2015, 2017 and clang).

drdanz, 2018-05-17:
Fixed by #1647

closed 2018-05-17 by drdanz