robotology/yarp #1580: YARP version after 3.0.0

drdanz, 2018-03-09:
I'd like to re-discuss the versioning after the next feature release that will be YARP `3.0.0`

In the last few years, we used a 4 numbers versioning (i.e. `2.3.72` feature release, `2.3.72.1` bugfix release).
With the `3.0.0` release I would like to switch to a 3 number versioning.
This means that the following "bugfix" release version will be `3.0.1` (not `3.0.0.1`).

For the feature releases, we are currently using an even-odd numbering, that is a very popular approach if you live in the 90ies.
I'd like to propose to abandon the even-odd numbering and to make the following "feature" release `3.1.0`

To achieve this, and still be able to rely on new features on the devel branch, we should use the 3rd number to detect that we are using the devel branch. This can be achieved by using a "large" number, for example:

* `3.0.50`: Arbitrary big number increased manually when a new feature is added
* `3.0.20180309`: Use 8 digits date as "huge number". It can be updated manually or, eventually by a bot whenever there is a new commit.

What do you think? Please vote using reactions:

| reaction   | `devel`        | bugfix release | feature release | Comment                                                  |
|------------|----------------|----------------|-----------------|----------------------------------------------------------|
| :-1:       | `3.0.1.0`      | `3.0.0.1`      | `3.0.2`         | even-odd + 4 version numbers (current versioning system) |
| :confused: | `3.1.0`        | `3.0.1`        | `3.2.0`         | even-odd + 3 version number                              |
| :+1:       | `3.0.50`       | `3.0.1`        | `3.1.0`         | `devel` identified by arbitrary large number             |
| :heart:    | `3.0.20180309` | `3.0.1`        | `3.1.0`         | `devel` identified by date                               |

CC @robotology/all-robotology-developers  since YARP versioning will probably impact the version for all robotology-related software.

(related issue: #640)

alecive, 2018-03-09:
This election is fraudulent! You are biasing the vote because of your choice of the emojis :grin: 

Minor comment: instead of `3.0.20180309`, why not removing the `2` and the `0`? It would be a slimmer `3.0.180309` :sunglasses:

pattacini, 2018-03-09:
For an anonymous voting system: https://github.com/apex/gh-polls 😉

drdanz, 2018-03-09:
> This election is fraudulent! You are biasing the vote because of your choice of the emojis

You got me! Guess which one I voted? :laughing:

drdanz, 2018-03-09:
> For an anonymous voting system: https://github.com/apex/gh-polls

It's not supposed to be anonymous, each vote will count for the number of robochlorians of the user! :robot: 
Therefore @traversaro won! :rofl:

alecive, 2018-03-09:
> You got me! Guess which one I voted?

lol

> It's not supposed to be anonymous, each vote will count for the number of robochlorians of the user! Therefore @traversaro won!

That's correct, otherwise I would wake up the dead to make option :+1: win :sunglasses:

diegoferigo, 2018-03-26:
In my opinion the only meaningful versioning for devel would be the :heart: option, with the variant of `3.0.180309`. Using an arbitrary large number makes no sense at first look. Other projects also use the partial SHA of the last commit, such as `3.0.c7e9f32` as the time of writing, which I like equally.

drdanz, 2018-03-26:
The sha1 is an useful information for debugging issues (we do have that information in the "full" version), but the problem that it is not a monotonic function, therefore you cannot use that to ensure that a specific feature is available...

pattacini, 2018-03-26:
When voting, remember that hearts can be ... broken 💔 
😛

traversaro, 2018-03-26:
> In my opinion the only meaningful versioning for devel would be the heart option, with the variant of 3.0.180309. Using an arbitrary large number makes no sense at first look. 

The rationale of "arbitrary large number" is to have meaningful `find_package` checks on devel versions. Depending on a specific feature added in the devel branch may seem a corner-case use-case, but it is indeed something that we do **a lot**, at least in IIT.

drdanz, 2018-03-29:
:heart: and :+1: are very close, I suggest we take the date, but divided in half, so that we have an "arbitrary large number" that is at the same time nonsense, but at the same time related to the date. This way everyone should be happy! :rofl: 

Keep voting!

lornat75, 2018-03-29:
I think @traversaro has a good point.

drdanz, 2018-04-18:
Time to close this poll!
The "arbitrary large number (:+1:)" option wins, but we discussed internally and instead of `50`, we will use `100`. Therefore after YARP 3.0.0 release

|      | `devel`   | next bugfix release | next feature release |
|------|-----------|---------------------|----------------------|
| :+1: | `3.0.100` | `3.0.1`             | `3.1.0`              |

closed 2018-04-18 by drdanz