robotology/yarp #1581: Setup CLA bot

drdanz, 2018-03-09:
As discussed, we are planning to enable a the CLA assistant bot (https://cla-assistant.io/) as for 

 - [ ] Write the CLA content
 - [ ] Setup the CLA assistant

Is is possible to share the same CLA between several repositories, so that people has to sign it only once?

@claudiofantacci @traversaro @lornat75 @giorgiometta

drdanz, 2018-03-09:
> Is is possible to share the same CLA between several repositories, so that people has to sign it only once?

I think it is possible, but the checkbox "Share the Gist" should be ticked when adding the license, and I think it wasn't enabled for iDynTree. @traversaro

drdanz, 2018-03-09:
This is the current CLA for iDynTree

https://gist.github.com/traversaro/7bb08ed6eb3ca15d141752020e51ceb8

A couple of comments:

1. It should not be under one user gist, since the owner could decide to remove it any time as soon as he leaves IIT. Unfortunately it is not possible to have gists under an organization, therefore I suggest we create a "robotology-admin" user who will own the gist.

2. There should be a version in the agreement, for example

### Fondazione Istituto Italiano di Tecnologia Individual Contributor License Agreement
##### Version 1, 09 March 2018

3. It is not clear the relationship between "IIT" and "robotology". Does IIT own robotology? Does IIT own only a few robotology repositories? If someone outside IIT creates a repo and he wants it to included in robotology, does he have to give up copyright to IIT? This should be clarified.

lornat75, 2018-03-09:
Issue moved to [robotology/confidential-issues #1](https://github.com/robotology/confidential-issues/issues/1) via [**ZenHub**](https://www.zenhub.com/)

closed 2018-03-09 by drdanz