robotology/yarp #1584: found a bug in yarpscope

serena-ivaldi, 2018-03-11:
I found a bug in yarpscope. We wanted to visualize different yarpscope, launched from terminal (not with XML file). By default, the SimpleLoader class by default is creating a local port called /yarpscope. There is no way to change it with a local port name. So if I launch yarpscope on two terminals, there is a conflict because the second process wants to create a port that already exist.
This causes a second bug, which is you cannot have two scopes from the same port, which is what we wanted to do, since we wanted to monitor the force/torque sensors with two scopes, one for the forces and one for the torques.

I fixed the code of the yarpscope in the SimpleLoader to enable different port numbers. Can you add me to the developers so I can push it in a branch and do a pull request?

traversaro, 2018-03-11:
> I fixed the code of the yarpscope in the SimpleLoader to enable different port names. Can you add me to the developers so I can push it in a branch and do a pull request?

Hi Serena! The usual workflow in this case is that you create a **fork** of the yarp repository in your personal profile, and you push your branch there, so you can open a pull request without the need of having write access to the `robotology/yarp` repository. 
See:
* https://help.github.com/articles/fork-a-repo/ 
* https://help.github.com/articles/creating-a-pull-request-from-a-fork/

drdanz, 2018-03-11:
I think this was originally done on purpose, even though I cannot remember a valid reason why it should be like this...
The *workaround* normally used is to set the `YARP_PORT_PREFIX` environment variable.