robotology/yarp #1585: Include examples in the main build

drdanz, 2018-03-12:
Many examples are outdated and probably no longer build.
I think that we should add them to the main build (with some flag `YARP_COMPILE_EXAMPLES` disabled by default), so that we can easily notice when some of them should be updated.

traversaro, 2018-03-12:
Instead of adding them to the main build, adding them as a separate project, and just adding in Travis/AppVeyor/Jenkins the code to build them externally would ensure that the YARP CMake config machinery is always working.

drdanz, 2020-07-21:
Another big problem is that the documentation contains full examples inside the doxygen files. Instead they should include files from examples, so that they are easier to test and we can ensure that they are updated.

PR #2326 (2020-07-27): Build examples in main build
### Build System

* It is now possible to build some of the examples in the main build using the
  `YARP_COMPILE_EXAMPLES` CMake Option
* It is now possible to build some of the YARP examples as test using the
  `YARP_ENABLE_EXAMPLES_AS_TESTS` CMake Option

Fixes #1585
closed 2020-07-27 by drdanz