robotology/yarp #1588: Rename YARP_OS -> YARP_os

drdanz, 2018-03-12:
All the libraries have the name lowercase, but for historical reasons, YARP_OS is uppercase.
This might be a good moment to change it, therefore I suggest to rename the library to YARP_os, and to add ALIAS targets for YARP_OS.

As an alternate option (I'm not really sure if this can be done), we could have YARP_os as a normal library, and YARP_OS as an library that links both YARP_os and YARP_init.

Comments?

traversaro, 2018-03-12:
My secret CMake dream is to have `ALIAS` target that can generate a warning when linked, to deprecate easily old target names.

drdanz, 2018-03-13:
I totally agree and was thinking exactly the same thing while I was opening this issue.
Opened a related feature request in CMake https://gitlab.kitware.com/cmake/cmake/issues/17824 :grin:

closed 2020-12-22 by drdanz