robotology/yarp #1592: Hardcoded carrier names

drdanz, 2018-03-14:
YARP is supposed to be agnostic about which carrier is currently used, but in some places the carrier name is checked. For example:

https://github.com/robotology/yarp/blob/290545d37e59d5d8ae3f1c55df151c656b308152/src/libYARP_OS/src/NameServer.cpp#L367-L377

In this example `local` and `shmem` are checked explicitly, but this will not work for any other carrier.
I think we should add 2 methods to the `Carrier` interface that allows to perform this check, something like `sameProcessOnly()`, and `sameSystemOnly()` both returning to false if not explicitly implemented in the carrier (Any suggestion on the naming is welcome).

I'm not really sure if this is ok though, since this part will run (I think) on the yarpserver, and it could be running on a machine where the carrier is not available. For example imagine a that we had a carrier that runs only on windows, and the yarp server is running on linux... How can we handle this?

There are other hardcoded carrier names (mostly `tcp`), we should also investigate those.

Nicogene, 2018-03-15:
Unfortunately there is not the :nauseated_face: reaction. :grimacing: 

I totally agree to remove these hardcoded checks and define functions to the carrier level. 

In another carrier  that I worked to, I do this check inside the carrier itself during the initial handshake.
So if two ports are not on the same machine the connect fails, I don't know if it is possible to do the same for the `local` and `shmem` carrier.

lornat75, 2018-03-26:
what about:

`interProcess()`/`intraProcess()` and `interHost()`/`intraHost()`

drdanz, 2018-03-26:
Sounds good, but I always have doubts about `intra` vs `inter` :persevere:

drdanz, 2018-03-26:
After a quick search, I'd say that `intraProcess()` and `intraHost()` are the right ones...

lornat75, 2018-03-26:
if you don't need to further differentiate between `local` and `shmem` should be ok.

drdanz, 2018-03-26:
I think we should also specify in the documentation the exact meaning of these, because they are already confusing to me...

`intraProcess()` should be "works intra process only" (i.e. only `local`) or "works intra process" (i.e. all carriers)
`intraHost()` should be "works intra host only" (i.e. `shmem`) or "works intra host" (i.e. all carriers, except `local`)

lornat75, 2018-03-26:
Right. Is there the need to also specify that a protocol is `interHost` ?

drdanz, 2018-03-26:
```c++
/**
 * Returns true if this carrier works only between 2 ports on the same process
 */
bool interProcessOnly() = 0; // { return false; } in Abstract carrier, reimplemented in local carrier

/**
 * Returns true if this carrier works only between 2 ports on the same host
 */
bool interHostOnly() = 0; // { return false; } in Abstract carrier, reimplemented in shmem carrier
```

Or perhaps something like this would be better

```c++
enum class Foo
{
    None = 0,
    IntraProcess,
    IntraHost,
    Any
};

Foo aSignificantMethodName()
{
    return Foo::Any;
}
```

or also (over-engineering?)

```c++
enum class Foo
{
    None = 0,
    WorksIntraProcess = 0x0001,
    WorksIntraHost = 0x0002,
    WorksInterHost = 0x0004,

    IntraProcessOnly = WorksIntraProcess,
    IntraHostOnly = WorksIntraProcess | WorksIntraHost,
    Any = WorksIntraProcess | WorksIntraHost | WorksInterHost
};

Foo aSignificantMethodName()
{
    return Foo::Any;
}
```

barbalberto, 2018-03-27:
Personally, I'd prefer the first option. More clear and I get directly the info I want.
Btw, I think the name should be `intraXXX` instead of `interXXX`.

> Inter- is a prefix that means between two groups, and intra- is a prefix which means within or inside one group.

lornat75, 2018-03-27:
The second option has the advantage that it makes the options mutually exclusive, whereas in the first case you have to explicitly make sure that a carrier does not return true on both.

barbalberto, 2018-03-27:
I was wandering exactly about that.
A intra-host carrier could also be intra-process, in theory, while it is not true the other way around.

drdanz, 2018-03-27:
As discussed with @lornat75, we are proposing to use this naming:

```c++
enum class Foo
{
    None = 0,
    
    WorksIntraProcess = 0x0001,
    WorksIntraHost    = 0x0002,
    WorksInterHost    = 0x0004,

    IntraProcess = WorksIntraProcess
    IntraHostOnly = WorksIntraHost,
    IntraHost = WorksIntraProcess | WorksIntraHost,
    InterHostOnly = WorksInterHost,
    InterHost =  WorksIntraProcess | WorksIntraHost | WorksInterHost
};

closed 2020-12-22 by drdanz