robotology/yarp #1595: Thread and RateThread need a Network yarp object

jgvictores, 2018-03-15:
`RateThread` needs a `Network` yarp object, or else segfaults. This is replicable via the example ([doxy](http://www.yarp.it/os_2ratethread_8cpp-example.html), [GitHub](https://github.com/robotology/yarp/blob/290545d37e59d5d8ae3f1c55df151c656b308152/example/os/ratethread.cpp)).

I also ran the [thread example](https://github.com/robotology/yarp/blob/290545d37e59d5d8ae3f1c55df151c656b308152/example/os/threads.cpp), also segfaults.

Possibly related `bool useSystemClock`, but `initWithSystemClock()` looks pretty private, and in fact having to call manually it would look pretty hackish anyway.

There is WIP such as #1207, so not sure if this could be added as a sub-task there.

diegoferigo, 2018-03-15:
This is probably related to the big clock refactoring @barbalberto did in https://github.com/robotology/yarp/pull/1277. Tests have never been updated after that and the usage of `RateThread` with the new modifications might be different wrt the current tests status.

barbalberto, 2018-03-15:
RateThread requires Network for many reasons, the main one is that it uses Mutex which in turns requires Network to be initialized, and yes it is a bit counter intuitive.

I created a PR to fix the example, but the requirement for the RateThread to have Network initialized cannot be removed.

Nicogene, 2018-03-15:
Example fixed in #1596 .
I'm afraid that the dependency of `RateThread` to the `Network` object is intentional, so I would not define it as a bug. (I don't like it either)

jgvictores, 2018-03-15:
Just pending #1596 merge i guess. : )

drdanz, 2018-03-15:
I think we should add a note in the documentation of each class that requires the `Network`.
In this case there is a very good reason, i.e. the rate thread runs either on the system clock, or on the network clock depending on the clock configured.
There is a `SystemRateThread` that is supposed to be running only on the system clock, that inherits from `RateThread`, perhaps this class could be modified in order to run even without the Network? CC @barbalberto

jgvictores, 2018-03-15:
Maybe some checks can be performed... a segfault feels a bit painful given the known scenario.

francesco-romano, 2018-03-16:
Just remember that `Network` is an unfortunate name that actually initialises all the static objects. It is not entirely `Network` related. (actually if I recall correctly there is almost no network logic there)

barbalberto, 2018-03-26:
> There is a `SystemRateThread` that is supposed to be running only on the system clock, that inherits from RateThread, perhaps this class could be modified in order to run even without the Network? CC @barbalberto

I think `SystemRateThread` could already live without the `Network`, the point is that `SystemRateThread` uses  `mutex`, which requires `Network`, so I cannot test it.
It is the `Time` class the one that needs to be able to read from a port, so a `SystemRateThread` which calls only `SystemClock::nowSystem()` shall be fine.

> Just remember that Network is an unfortunate name that actually initialises all the static objects. It is not entirely Network related. (actually if I recall correctly there is almost no network logic there)

I agree ... a better name would be `yarp::os::Framework` :smile:

Nicogene, 2018-06-07:
I tryed the [threads example](https://github.com/robotology/yarp/blob/devel/example/os/threads.cpp) and [ratethread example](https://github.com/robotology/yarp/blob/devel/example/os/ratethread.cpp) removing the `Network` object and after #1718 they work without problems :muscle: 
I get only this warning run-time:
```
yarp: 
 Warning an issue has been found, please update the code.
     Clock is not initialized: This means YARP framework has not been properly initialized. 
     The clock can be initialized with one of the following methods:
     - Create yarp::os::Network object or call yarp::os::Network::init()
     - Call useSystemClock()
     otherwise use yarp::os::SystemClock::nowSystem() and yarp::os::SystemClock::delaySystem() instead of Time::now() and Time::delay()

```

closed 2018-06-07 by Nicogene

barbalberto, 2018-06-07:
It may work, but that's not guaranteed.
As stated in the yarp council, every program that uses yarp classes MUST initialize the `Network` object beforehand.
So the only correct answer to this and #1718 issue is: yes it is required. Not doing so it will create undefined behaviour.