robotology/yarp #1602: Operators should not be in the yarp::math namespace

drdanz, 2018-03-15:
`Vector` and `Matrix` operators are defined inside the `yarp::math` namespace.

https://github.com/robotology/yarp/blob/29e9108a52713dbe737d9a9c44a62da3adc3ce51/src/libYARP_math/include/yarp/math/Math.h#L16-L27

This means that, in order to perform an operation (for example `Vector b = a + s;`), the user has to add `using namespace yarp::math;` or to call the operator explicitly, i.e. (I think) something like this: `Vector b = yarp::math::operator+(a, s);`
The first option is considered bad practice, and the second option is just wrong.

I think we could just move them outside of the namespace, the only effect is if someone is using the second syntax, but I don't think anyone would be doing that...

drdanz, 2018-03-15:
By the way, why are they in math? I thought they were using gsl/eigen, but most of them are not...

https://github.com/robotology/yarp/blob/29e9108a52713dbe737d9a9c44a62da3adc3ce51/src/libYARP_math/src/math.cpp#L24-L148

lornat75, 2018-03-15:
I am ok moving the operators outside yarp::math. I would keep however the current implementation when it relies on linear algebra lib, and keep them on libYARP_math.

claudiofantacci, 2018-03-16:
I would be super happy to see math operations moved outside from `yarp::math`.

We could take the chance to update the implementation (if possible, but I don't see any problem in doing that) of those operations that are not implemented with `Eigen`.

jgvictores, 2018-03-17:
Has certain implications on #1603 and #1559.

Moreover, where do they go? `yarp::sig::` ...maybe simply `yarp::`?

lornat75, 2018-03-19:
My understanding was that the operators would be moved outside any namespace. From what I know it is the only way to solve the problem raised by @drdanz.

pattacini, 2018-03-19:
@drdanz said
>I think we could just move them outside of the namespace, the only effect is if someone is using the second syntax (i.e. `Vector b = yarp::math::operator+(a, s);`) but I don't think anyone would be doing that...

I'm not aware of anyone resorting to the second syntax, but, actually, there's the risk it is being used anywhere in header files, that is where declaring `using namespace yarp::math` is even a worse practice. Then, using math in header files can be considered questionable, I know, but I just wanted to point out this possible use case.

PeterBowman, 2018-03-20:
> the user has to add using namespace yarp::math; or to call the operator explicitly

Somewhere between:

```c++
#include <iostream>

#include <yarp/sig/Vector.h>
#include <yarp/math/Math.h>

int main()
{
    yarp::sig::Vector v1, v2;
    v1.push_back(1);
    v1.push_back(2);
    v2.push_back(3);
    v2.push_back(4);
    
    using yarp::math::operator+;
    yarp::sig::Vector v3 = v1 + v2;
    
    std::cout << "[" << v3[0] << ", " << v3[1] << "]" << std::endl;
    
    return 0;
}
```

Output: `[4, 6]`.

drdanz, 2018-03-22:
> [...]
```c++
using yarp::math::operator+;
```

Right, but still ugly...

jgvictores, 2018-04-05:
> My understanding was that the operators would be moved outside any namespace.

After a close encounter with a [Pythonic error](https://stackoverflow.com/questions/38730671/error-with-pow-cant-convert-complex-to-float), where at least two different math implementations exist, and forcing one of them into the "global namespace" makes the same program line break at runtime:
```python
>>> pow(1+1j, 1)
(1+1j)
>>> from math import *
>>> pow(1+1j,2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert complex to float
```
I'm definitely more inclined to keep a the status quo, promoting the proposal by @PeterBowman above:
```cpp
{
using yarp::math::operator+;
yarp::sig::Vector v3 = v1 + v2;
}
```
Additional/related benefits:
1. Different internal implementations of the same operation, e.g. `gsl` vs `eigen` vs `pcl` implementations for matrix multiplication (in line with [this comment](https://github.com/robotology/yarp/issues/1603#issuecomment-374756907) by @pattacini on `math` implementations: `available, yet-to-come, won't-provide`), maintaining flexibility/modularitity.
2. Different operations implemented in completely different use cases that still use the same types, e.g. a matrix `yarp::morphology::operator+` providing dilatation (perhaps a bit far-fetched, but insert your more robotic examples here, xD).

lornat75, 2018-04-05:
@jgvictores in the case of YARP potential clashes would only be with YARP types, in case somebody wants to provide his own set of operators. In this case it would be enough to NOT include headers from libYARP_math.

jgvictores, 2018-04-07:
Please PM me if this itch is bugging too much, no intention to start a :fire:-war over this or any other topic. I'm a big fan of retracting my comments. :joy:

Just a fun guessing game: If `using namespace yarp::math` is bad practice... then hard-coding the equivalent to including this line within the math headers is...

lornat75, 2018-04-09:
@jgvictores I see your point. 

I have no strong opinions, when I first added the operators in the `math` namespace I decided to follow the strict guidelines to have all functions within the namespace. I must have considered (it has been a long time now) that it was not a big deal limiting the scope of the `using` directive to c/cpp files and inline functions in header files.

However, it seems also acceptable to have operators in the global namespace, if this makes many people happier...

jgvictores, 2018-04-10:
I guess I'm just worried about establishing a precedent for bad practices. Even very good projects can get contaminated by ill-posed design decisions, and later suffer consequences such as breaking changes as-a-habit.

Perhaps introducing a `yarp::math::operators` namespace for lazy/q&d programmers to include a `using namespace yarp::math::operators;` line would be a sensible solution?

Nicogene, 2018-05-25:
Closed (for now) by #1700

PR #1700 (2018-05-25): move math operators outside yarp::math namespace
fixes #1602
closed 2018-05-25 by Nicogene