robotology/yarp #1605: Compilation problem in devel branch with LUA bindings

GiulioRomualdi, 2018-03-16:
When I try to compile the `devel` branch the following error occurs
```make
make[2]: *** No rule to make target '../src/libYARP_dev/include/yarp/dev/IKinectDeviceDriver.h', needed by 'bindings/lua/yarpLUA_wrap.cxx'.  Stop.
CMakeFiles/Makefile2:5812: recipe for target 'bindings/lua/CMakeFiles/yarp_lua.dir/all' failed
make[1]: *** [bindings/lua/CMakeFiles/yarp_lua.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```
cc @S-Dafarra

Nicogene, 2018-03-16:
```
$ cd $YARP_DIR
$ rm -rf ./bindings
$ cmake ..
$ make
```
Should be the solution of your problem.
Recently several header of `YARP_dev` have been moved and it seems you have some problems in the makefiles.

Try it and let me know!

GiulioRomualdi, 2018-03-16:
Solved! Thank you!

closed 2018-03-16 by Nicogene

traversaro, 2018-03-16:
See https://github.com/robotology/idyntree/issues/84 for the more general error related to this.