robotology/yarp #1607: Improvements in bindings (automatically tag all the functions that need type conversion)

jgvictores, 2018-03-16:
Copied off comment by @David-Estevez ([ref](https://github.com/roboticslab-uc3m/yarp-devices/issues/145#issuecomment-348946216)), a long-shot:

Reading the [SWIG docs](http://www.swig.org/Doc1.3/Arguments.html), there seems to be a way to automatically tag all the functions that need type conversion due to the pointers with a function that performs that conversion:

```cpp
%module example
%include "typemaps.i"

%apply double *OUTPUT { double *result };

%inline %{
extern void add(double a, double b, double *result);
extern void sub(double a, double b, double *result);
extern void mul(double a, double b, double *result);
extern void div(double a, double b, double *result);
%}
```

And this works even with several return values:

```cpp
%include "typemaps.i"
%apply int *OUTPUT { int *width, int *height };

// Returns a pair (width,height)
void getwinsize(int winid, int *width, int *height);
```

Which gives this output:
``` 
>>> w,h = genwinsize(wid)
>>> print w
400
>>> print h
300
>>>
```

We could possibly dramatically reduce the size of `yarp.i` (>1400 lines at time of writing) while gaining features.

PeterBowman, 2020-06-27:
Still, you cannot avoid pointer handling aimed for interpreting inputs/outputs as arrays. For instance:

https://github.com/robotology/yarp/blob/f91ceb574b133b0cd45981b959d536dd62ac2e5e/bindings/yarp.i#L1013-L1015

This rewritten method corresponds to the original `getControlModes(int * modes)` signature. Unless there is a specific SWIG construct to handle these cases, they shall not be removed from *yarp.i* nor simplified.

PeterBowman, 2020-06-29:
I have been playing a bit with this feature and concluded that size reduction is not quite possible in the best case, and in the worse the whole pointer argument deduction is not attainable at all. To sum up, we currently do this:

```c++
%{
#include <yarp/dev/IEncoders.h>
%}

// pick definition of yarp::dev::IEncoders
%include <yarp/dev/IEncoders.h>

// add new methods/signatures to the default interface
%extend yarp::dev::IEncoders {
    int getAxes() {
        int buffer;
        bool ok = self->getAxes(&buffer);
        if (!ok) return 0;
        return buffer;
    }

    // here go *only* those methods which really need proper pointer argument handling
}
```

Now, in order to take advantage from the SWIG [feature](http://www.swig.org/Doc3.0/Arguments.html) discussed here:

```c++
%include "typemaps.i"

%{
#include <yarp/dev/IEncoders.h>
%}

namespace yarp {
namespace dev {

// define this class from scratch
class IEncoders {
public:
    bool getAxes(int * OUTPUT); // no definition needed

    // *all* remaining methods
};

} // namespace dev
} // namespace yarp
```

Note this is just an approximation that works with simpler classes. `IEncoders` et al. are abstract interfaces, which makes things more complicated (perhaps even impossible to achieve).