robotology/yarp #1614: Using YARP devel from a C++14 codebase result in error related to YARP_OS_DEPRECATED_API

gabrielenava, 2018-03-22:
This may seem a duplicate of https://github.com/robotology/yarp/issues/1531 but this results in a compilation error. 
```
In file included from /home/gnava/Software/gitHub/robotology/robotology-superbuild/build/install/include/yarp/os/api.h:13:0,
                 from /home/gnava/Software/gitHub/robotology/robotology-superbuild/build/install/include/yarp/os/ConstString.h:13,
                 from /home/gnava/Software/gitHub/robotology/robotology-superbuild/build/install/include/yarp/os/ContactStyle.h:13,
                 from /home/gnava/Software/gitHub/robotology/robotology-superbuild/build/install/include/yarp/os/Network.h:13,
                 from /home/gnava/Software/gitHub/robotology/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/camera/include/gazebo/Camera.hh:13,
                 from /home/gnava/Software/gitHub/robotology/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/camera/src/Camera.cc:8:
/home/gnava/Software/gitHub/robotology/robotology-superbuild/build/install/include/yarp/os/Stamp.h:96:7: error: expected identifier before ‘__attribute__’
 class YARP_OS_DEPRECATED_API Stamped
       ^
In file included from /home/gnava/Software/gitHub/robotology/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/camera/include/yarp/dev/CameraDriver.h:12:0,
                 from /home/gnava/Software/gitHub/robotology/robotology-superbuild/robotology/GazeboYARPPlugins/plugins/camera/src/Camera.cc:9:
/home/gnava/Software/gitHub/robotology/robotology-superbuild/build/install/include/yarp/os/Stamp.h:98:1: error: expected primary-expression before ‘public’
 public:
 ^
/home/gnava/Software/gitHub/robotology/robotology-superbuild/build/install/include/yarp/os/Stamp.h:98:1: error: expected ‘}’ before ‘public’
/home/gnava/Software/gitHub/robotology/robotology-superbuild/build/install/include/yarp/os/Stamp.h:98:1: error: expected ‘,’ or ‘;’ before ‘public’
/home/gnava/Software/gitHub/robotology/robotology-superbuild/build/install/include/yarp/os/Stamp.h:108:30: error: ‘virtual’ outside class declaration
     virtual Stamp getStamp() const = 0;
                              ^
/home/gnava/Software/gitHub/robotology/robotology-superbuild/build/install/include/yarp/os/Stamp.h:108:30: error: non-member function ‘yarp::os::Stamp yarp::os::getStamp()’ cannot have cv-qualifier
/home/gnava/Software/gitHub/robotology/robotology-superbuild/build/install/include/yarp/os/Stamp.h:108:38: error: function ‘yarp::os::Stamp yarp::os::getStamp()’ is initialized like a variable
     virtual Stamp getStamp() const = 0;
                                      ^
/home/gnava/Software/gitHub/robotology/robotology-superbuild/build/install/include/yarp/os/Stamp.h:112:1: error: expected declaration before ‘}’ token
 } // namespace yarp
 ^
plugins/camera/CMakeFiles/gazebo_yarp_camera.dir/build.make:62: recipe for target 'plugins/camera/CMakeFiles/gazebo_yarp_camera.dir/src/Camera.cc.o' failed
make[5]: *** [plugins/camera/CMakeFiles/gazebo_yarp_camera.dir/src/Camera.cc.o] Error 1
CMakeFiles/Makefile2:408: recipe for target 'plugins/camera/CMakeFiles/gazebo_yarp_camera.dir/all' failed
make[4]: *** [plugins/camera/CMakeFiles/gazebo_yarp_camera.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/GazeboYARPPlugins.dir/build.make:107: recipe for target 'robotology/GazeboYARPPlugins/CMakeFiles/YCMStamp/GazeboYARPPlugins-build' failed
make[2]: *** [robotology/GazeboYARPPlugins/CMakeFiles/YCMStamp/GazeboYARPPlugins-build] Error 2
CMakeFiles/Makefile2:1882: recipe for target 'CMakeFiles/GazeboYARPPlugins.dir/all' failed
make[1]: *** [CMakeFiles/GazeboYARPPlugins.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2
```
This is happening with ``gazebo-yarp-plugins`` and other projects requiring C++14.

Usually it is enough to remove ``YARP_OS_DEPRECATED_API`` from Stamp.h

cc @S-Dafarra @traversaro

traversaro, 2018-03-22:
I guess this is a problem of compiling with Gazebo9 that requires C++14.
The problem is actually this one: https://stackoverflow.com/questions/40886628/deprecated-attribute-visibility-default-in-gcc-6-2 . 
`YARP_DEPRECATED` uses the C++14's style `[[deprecated]]`, but `YARP_OS_API` uses the c++98/11 gcc extention `__attribute__ ((visibility ("default")))`

traversaro, 2018-03-22:
Solution provided in https://github.com/robotology/yarp/issues/1614 .

traversaro, 2018-03-22:
I just realized that all the other projects are not having this problem because they rely on the deprecation macro provided by the `generate_export_header` macro, that is not defining the deprecated macro as `[[deprecated]]` even if the C++ standard is C++14 .

drdanz, 2018-03-23:
> Solution provided in #1614 .

I think this should be #1615

closed 2018-04-11 by drdanz