robotology/yarp #1620: BufferedPort API read() and prepare() return different types

lornat75, 2018-03-23:
`BufferedPort::read()` returns a pointer, while `BufferePort::prepare()` returns a reference. I think it would be more appropriate if `prepare()` returned a pointer, if anything, for symmetry. Given that C++ does not allow overriding functions with different return values, we could deprecate `prepare()` and add a new function e.g. `prepareWrite()` that returns a pointer.

@claudiofantacci suggested to return a shared smart pointers, but I am not sure we are ready for that.

I understand this is a minor issue, but given we are revising the YARP API it could be a good time to do it. If however others think it is not fundamental (or there are reasons I do not see that makes returning a reference more appropriate)  I can live with that.

pattacini, 2018-03-23:
I think it's appropriate that `prepare()` does return a reference rather a pointer. This way, it's clear whose the ownership is and we don't have to check against `nullptr`, given that it's always allocated.

In practice, the only reason why `read()` gives back a pointer is to let the user check whether there's a new message available.

I'm also against using a _stray_ smart pointer in the API, while the rest is based on raw pointers.

drdanz, 2018-03-23:
I agree with @pattacini. Also returning a raw pointer is not considered a good practice.

> I'm also against using a *stray* smart pointer in the API, while the rest is based on raw pointers.

For now, I agree, but if, at some point, we should decide to use the smart pointers, we will have to start somewhere. For example, in this case, one day I'd like to change `BufferedPort<T>::read()` to return a `shared_ptr<T>` or a `weak_ptr<T>`. At this point we could also consider changing `prepare` to return a smart pointer, in order to have a symmetric API.

pattacini, 2018-03-23:
- 👍 for refactoring in favor of smart pointers.
- 👍 for an exhaustive refactoring that will take place once for all.
- 👎 for incremental refactoring that will force us to cycle over and over through the immense amount of code spread around.
- 🤔 although it's true the smart pointers resolve ownership, I'm still not 100% sure that a smart pointer is preferable over a reference when the latter is applicable.

lornat75, 2018-03-23:
This was not intended to escalate so much, let's close it before it becomes a huge issue :)

closed 2018-03-23 by lornat75