robotology/yarp #1626: yarp::os::PortablePair crash when using multiple readers

randaz81, 2018-03-26:
The following snippet crashes (in a random time <10 seconds) if two `yarp read ... /zzz` are executed simultaneously.

```
int main(int argc, char *argv[])
{
    Network yarp;
    typedef yarp::os::PortablePair <yarp::os::Bottle, yarp::os::Bottle> CommandMessage2;
    yarp::os::BufferedPort<CommandMessage2> p;
    p.open("/zzz");
    while (1)
    {
        CommandMessage2& b = p.prepare();
        b.head.clear();
        b.body.clear();
        b.head.addDouble(1.0);
        b.body.addDouble(1.0);
        p.write();
    }
   return 0;
}
```

The following code, instead, is not affected by this issues. This suggests the fact that the problem **may** reside in the `yarp::os::PortablePair`.

```
int main(int argc, char *argv[])
{
    Network yarp;
    yarp::os::BufferedPort<yarp::os::Bottle> p;
    p.open("/zzz");
    while (1)
    {
        Bottle& b =p.prepare();
        b.clear();
        Bottle& b1 = b.addList();
        Bottle& b2 = b.addList();
        b1.addVocab(VOCAB_VELOCITY_MOVE);
        b1.addInt(2.0);
        b2.addDouble(4.0);
        p.write();
    }
   return 0;
}
```

Please note that snippet 1 is inspired to RemoteControlBoard.cpp(*) which exhibits the same crash if, for example, two yarp readers are connected to `/<client_name>/command:o`. This should be thus considered a major bug. 

PS:(*) in RemoteControlBoard the following definition is used: ```typedef yarp::os::PortablePair <yarp::os::Bottle, yarp::sig::Vector> CommandMessage;```. In snippet 1, I replaced  yarp::sig::Vector with yarp::os::Bottle for simplicity.

lornat75, 2018-03-30:
@randaz81 do you mind uploading this example with CMake etc in a public repository so we can have a look?

pattacini, 2018-03-30:
@randaz81 you may want to use [Gist](https://gist.github.com) for that.
An example: https://gist.github.com/pattacini/6572864f8ec97c7f4639.

randaz81, 2026-11-05:
yarp::os::PortablePair class have been deprecated

closed 2026-11-05 by randaz81