robotology/yarp #1630: yard-config triggers homebrew warning

Nicogene, 2018-03-30:
Typing: `brew doctor` I get this warning:

```bash
Warning: "config" scripts exist outside your system or Homebrew directories.
`./configure` scripts often look for *-config scripts to determine if
software packages are installed, and what additional flags to use when
compiling and linking.

Having additional scripts in your path can confuse software installed via
Homebrew if the config script overrides a system or Homebrew provided
script of the same name. We found the following "config" scripts:
  /Users/nicologenesio/robotology/yarp/build/bin/Debug/yarp-config
```
@claudiofantacci @vtikha @traversaro can I safely ignore it or it is an issue?

Nicogene, 2018-03-30:
Yet Another Question, what is `yarp-config` ? 😅

pattacini, 2018-03-30:
Here's a link: [yarp-config](http://www.letmegooglethat.com/?q=yarp-config) 😛

claudiofantacci, 2018-04-05:
> can I safely ignore it or is it an issue?

You can safely ignore it.

A side note: `brew doctor` is a command used by `brew` maintainer to debug issues reported by the users. The output of the command does not strictly imply any error in your configuration.

Nicogene, 2018-04-05:
ok thanks for the tip :wink:

closed 2018-04-05 by Nicogene