robotology/yarp #1637: Question/proposal: Default to -DSKIP_ACE=ON for YARP 3 on Linux/Mac

jgvictores, 2018-04-06:
Here's a gutsy question/proposal: Default to -DSKIP_ACE=ON for YARP 3 on Linux/Mac

My personal opinion is that I really like the YARP support for Windows and would not like it to be dropped because of lack of support for ACE, which provides compatibility. However, depending on a library for no known reason, which is the Linux use case, does not seem efficient.

Just opening discussion here. What are your opinions?

jgvictores, 2018-04-07:
As said, it's just a proposal, perhaps even a thought experiment. :smiley: 

@drdanz Thanks a lot for looking into this, it's great to at least have a list of features that are expected/missing when using `-DSKIP_ACE=ON`.

Nicogene, 2018-04-07:
@drdanz I can check the points you exposed about macOS 😏

francesco-romano, 2018-04-09:
@drdanz macOS should work if the pthread implementation is Unix specific and not Linux specific
Same for dlopen/dlclose

drdanz, 2018-06-25:
About ACE for handling plugins, this part is currently being moved to https://github.com/robotology-playground/sharedlibpp, and it no longer requires ACE

closed 2020-12-22 by drdanz