robotology/yarp #1640: yarpserver --read is broken

claudiofantacci, 2018-04-10:
Tested on Windows, issuing
```bash
> yarp conf --clean
Cleared configuration file:
  C:\Users\cfantacci\AppData\Roaming\yarp\config\_claudio.conf
```

```bash
> yarp conf 172.17.20.49 20000
Configuration file:
  C:\Users\cfantacci\AppData\Roaming\yarp\config\_claudio.conf
Now stores:
  host 172.17.20.49 port number 20000 (yarp name server)
```

```bash
> yarpserver
    __  __ ___  ____   ____
    \ \/ //   ||  _ \ |  _ \
     \  // /| || |/ / | |/ /
     / // ___ ||  _ \ |  _/
    /_//_/  |_||_| \_\|_|
    ========================

Call with --help for information on available options
Using port database: :memory:
Using subscription database: :memory:
IP address: default
Port number: 10000
PROBLEM: need to change settings in C:\Users\cfantacci\AppData\Roaming\yarp\config\_claudio.conf
  Current settings: host 172.17.20.49 port 20000 family yarp
  Desired settings:  host 172.17.20.49 port 10000 family yarp
Please specify '--write' if it is ok to overwrite current settings, or
Please specify '--read' to use the current settings, or
delete C:\Users\cfantacci\AppData\Roaming\yarp\config\_claudio.conf
Aborting.
```

The following command should allow my custom setting to be used.
```bash
> yarpserver --read
    __  __ ___  ____   ____
    \ \/ //   ||  _ \ |  _ \
     \  // /| || |/ / | |/ /
     / // ___ ||  _ \ |  _/
    /_//_/  |_||_| \_\|_|
    ========================

Call with --help for information on available options
Using port database: :memory:
Using subscription database: :memory:
IP address: default
Port number: 10000
PROBLEM: need to change settings in C:\Users\cfantacci\AppData\Roaming\yarp\config\_claudio.conf
  Current settings: host 172.17.20.49 port 20000 family yarp
  Desired settings:  host 172.17.20.49 port 10000 family yarp
Please specify '--write' if it is ok to overwrite current settings, or
Please set an existing address in config file, or
delete C:\Users\cfantacci\AppData\Roaming\yarp\config\_claudio.conf
Aborting.
```

However the only option left to me is to use again `--write` that overwrites my custom setting.
```bash
> yarpserver --write
    __  __ ___  ____   ____
    \ \/ //   ||  _ \ |  _ \
     \  // /| || |/ / | |/ /
     / // ___ ||  _ \ |  _/
    /_//_/  |_||_| \_\|_|
    ========================

Call with --help for information on available options
Using port database: :memory:
Using subscription database: :memory:
IP address: default
Port number: 10000
yarp: Port /claudio active at tcp://172.17.20.49:10000/

Registering name server with itself:
 * register "/claudio" tcp "172.17.20.49" 10000
   + set "/claudio" offers http name_ser local tcp fast_tcp mcast udp text text_ack
   + set "/claudio" accepts http name_ser local tcp fast_tcp mcast udp text text_ack
   + set "/claudio" ips "127.0.0.1" "10.255.35.175" "172.17.20.49"
   + set "/claudio" process 16012
 * register fallback mcast "224.2.1.1" 10000
   + set fallback offers http name_ser local tcp fast_tcp mcast udp text text_ack
   + set fallback accepts http name_ser local tcp fast_tcp mcast udp text text_ack
   + set fallback ips "127.0.0.1" "10.255.35.175" "172.17.20.49"
   + set fallback process 16012
 * set "/claudio" nameserver "true"
Name server can be browsed at http://172.17.20.49:10000/

Ok.  Ready!
```

cc @pattacini @randaz81 @drdanz

Nicogene, 2018-04-10:
Have ever been **not** broken ? :sweat_smile: 
BTW the `--read` option is missing in `yarpserver --help`.

claudiofantacci, 2018-04-10:
The problem is that if you have two NIC you can't choose which one to use.

randaz81, 2018-04-10:
> Has ever been not broken ?

I think it was working on Mar 31 2016 😅

randaz81, 2018-04-10:
Bisect complete! (this one was really a looooong one!)
Issue found in commit  7ec99a6c73

barbalberto, 2018-04-11:
Wait guys, there is some misunderstanding here.

It looks the point is not about having more NIC, that part works. It complains about the yarpserver **_port_**, not IP address. For some reason the server does like to start on port != 10000

This test for example:
```
ifconfig 
enp0s31f6 Link encap:Ethernet  HWaddr d4:81:d7:78:65:5c  
          inet addr: 10.255.35.57  Bcast:10.255.35.255  Mask:255.255.255.0
          inet6 addr: fe80::8741:cf84:e090:65d1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1

yarp conf 10.255.35.57 10000 --> yarp server starts

yarp conf 10.255.35.57 10001  --> yarpserver does not start
```

If I configure the server to start on a different interface:
```
enp0s31f6 Link encap:Ethernet  HWaddr d4:81:d7:78:65:5c  
          inet addr:10.255.35.57  Bcast:10.255.35.255  Mask:255.255.255.0
          inet6 addr: fe80::8741:cf84:e090:65d1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:85226 errors:0 dropped:0 overruns:0 frame:0
          TX packets:20220 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:106868982 (106.8 MB)  TX bytes:2656409 (2.6 MB)
          Interrupt:16 Memory:e1200000-e1220000 

enp0s31f6:1 Link encap:Ethernet  HWaddr d4:81:d7:78:65:5c  
          inet addr:10.23.45.56  Bcast:10.255.255.255  Mask:255.0.0.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          Interrupt:16 Memory:e1200000-e1220000 

yarp conf 10.23.45.56  10000  --> it starts happily

 yarpserver
    __  __ ___  ____   ____
    \ \/ //   ||  _ \ |  _ \
     \  // /| || |/ / | |/ /
     / // ___ ||  _ \ |  _/
    /_//_/  |_||_| \_\|_|
    ========================

Call with --help for information on available options
Using port database: :memory:
Using subscription database: :memory:
IP address: default
Port number: 10000
yarp: Port /alby active at tcp://10.23.45.56:10000 

Registering name server with itself:
 * register "/alby" tcp "10.23.45.56" 10000
   + set "/alby" offers http name_ser local shmem tcp fast_tcp mcast udp text text_ack bayer mjpeg portmonitor priority tcpros rossrv xmlrpc
   + set "/alby" accepts http name_ser local shmem tcp fast_tcp mcast udp text text_ack bayer mjpeg portmonitor priority tcpros rossrv xmlrpc
   + set "/alby" ips "127.0.0.1" "10.255.35.57" "10.23.45.56" "::1" "fe80::8741:cf84:e090:65d1%2"
   + set "/alby" process 8099
 * register fallback mcast "224.2.1.1" 10000
   + set fallback offers http name_ser local shmem tcp fast_tcp mcast udp text text_ack bayer mjpeg portmonitor priority tcpros rossrv xmlrpc
   + set fallback accepts http name_ser local shmem tcp fast_tcp mcast udp text text_ack bayer mjpeg portmonitor priority tcpros rossrv xmlrpc
   + set fallback ips "127.0.0.1" "10.255.35.57" "10.23.45.56" "::1" "fe80::8741:cf84:e090:65d1%2"
   + set fallback process 8099
 * set "/alby" nameserver "true"
Name server can be browsed at http://10.23.45.56:10000/
```

but if I change the port 

```
yarp conf 10.23.45.56  10001

    __  __ ___  ____   ____
    \ \/ //   ||  _ \ |  _ \
     \  // /| || |/ / | |/ /
     / // ___ ||  _ \ |  _/
    /_//_/  |_||_| \_\|_|
    ========================

Call with --help for information on available options
Using port database: :memory:
Using subscription database: :memory:
IP address: default
Port number: 10000
PROBLEM: need to change settings in /home/icub/.config/yarp/_alby.conf
  Current settings: host 10.23.45.56 port _10001_ family yarp
  Desired settings:  host 10.23.45.56 port 10000 family yarp
Please specify '--write' if it is ok to overwrite current settings, or
Please specify '--read' to use the current settings, or
delete /home/icub/.config/yarp/_alby.conf
Aborting.
```

It does not. And the same behaviour is present before the commit 7ec99a6c73. Tested with commit 168942cf353

So the issue may be **yarpserver cannot start with port != 10000**, `yarpserver --read` has nothing to do with this behaviour.

randaz81, 2018-04-11:
Sorry, there is some misunderstanding here. And there are multiple issues.
In my tests I NEVER changed the port number. I always used port 10000 (despite what Claudio was saying at the beginning). And I can confirm that yarp server --read does not work with port 10000.

randaz81, 2018-04-11:
After f2f meeting, we clearly identified that we were speaking of two different issues.
The problem I was observing derives from the fix of issue #1404, that was not considering a very specific use case. I'm thus re-opening #1404 and continuing the discussion there.

randaz81, 2018-06-04:
commit 7ec99a6c73729 has been reverted to allow` yarp --read ` option even when the server ip address does not belong to the physical machine on which the server runs (e.g. when using DMZ, NAT, wifi networks...)

closed 2020-06-23 by drdanz