robotology/yarp #1645: Move libyarpc in bindings?

drdanz, 2018-04-13:
The `libyarpc` library is a sort of binding c for YARP.
At the moment it is considered unmaintained, but there are a few projects that use it, for example [YARP.jl](https://github.com/robotology-playground/YARP.jl/blob/d4e798bf6bbef9f0f37ebac86abbce221801c76c/src/YARP.jl#L14) (julia bindings for YARP).

I think it makes sense to go back to supporting it "officially", but I would like to move it in the `bindings/c` folder, even if it does not use swig. What do you think?

traversaro, 2018-04-16:
I agree moving `libyarpc` in `bindings`. 

Just for reference, there is a also a (currently not worked on) branch in SWIG that supports having C as a target language, see : https://github.com/swig/swig/issues/800 .

drdanz, 2018-04-17:
@traversaro if the branch is working, we might consider doing the same as for Matlab bindings...

closed 2020-12-22 by drdanz