robotology/yarp #1651: yarp::dev::FrameTransformClient::getTransform too verbose

xEnVrE, 2018-04-17:
When using `yarp::dev::FrameTransformClient::getTransform` it may happen that the required transform is not ready yet. If `getTransform`is used within a periodic module this produces a large numbers of Error messages 
`
FrameTransformClient::getTransform() frames <source_frame_id> and <target_frame_id> are not connected
`
which may be undesirable for the final user.

Since this is the only error condition handled by the method and already given by the return value of the method could it be an option to remove the `yError` messages?

traversaro, 2018-04-21:
@aerydna do you have an opinion on this?

randaz81, 2018-04-23:
Uhm, well, ROS displays this message, and I recognize it may be useful to identify the source issue when something in the system goes wrong.
What about putting it inside a `if(verbose)` ? (default false, value set by --verbose configuration option when the client is instantiated). BTW, there are several other methods what should be fixed in the same way.

aerydna, 2018-04-23:
IMHO the message should be displayed by the user application on receiving `false` when calling `getTransform` not by the TransformClient. In general, i think is better to leave the **user** comunication task to the **user** application , and not to entities in the middle.. BTW it is simply an opinion and if you prefer to implement a verbose mode it's ok for me