robotology/yarp #1665: yarpdev segmentation fault 11 on macOS

amaroyo, 2018-04-27:
Hello all, 

I am working on the **devel** branch (up to date), using a macOS (10.13.4). I am trying to run [svox](https://github.com/robotology/speech/tree/master/svox-speech) as written there. Actually, I am able to run the yarpdev module successfully, and change few settings with rpc. However, when I run the say command, there's a ```segmentation fault: 11``` in the yarpdev module. 

As far as I understood, in the new macOS, the [segmentation fault 11](https://discussions.apple.com/thread/8217579) is caused by an application attempting to access restricted memory space, and the operating system stopping this attempt.

I recompiled yarp, icub, speech and svox in Debug configuration, and also run it with --verbose but I couldn't get any more info.

Any ideas?

Thanks, 
Alex.

Nicogene, 2018-04-27:
issue moved robotology/speech#18

closed 2018-04-27 by Nicogene

pattacini, 2018-04-27:
Hi @amaroyo 
I'm only a mere user of speech/svox and not a Mac user.

Nicogene, 2018-04-27:
I never used svox but a mac user (newbie) but I can take a look :+1:
I'm agnostic about what this specific device does, but since `yarpdev` is simply a plugin loader, I think that the problem is in `svox`