robotology/yarp #1670: Module ignores `YARP_CLOCK` depending on where the Yarp Network is defined

aikolina, 2018-05-03:
I wanted to run a module in Gazebo and synchronize it through the Network clock instead of the System clock. To do so, I run Gazebo with the option `-slibgazebo_yarp_clock.so` and the module with `YARP_CLOCK=/clock`, but the module ignored completely this setting and used the System clock (the output of `isNetworkClock()` was always `false`).

After discussing with @traversaro and @pattacini, we found out that the Yarp Network was defined **after** some manipulation on the Resource Finder.
```C++
yarp::os::ResourceFinder &rf = yarp::os::ResourceFinder::getResourceFinderSingleton();
...
yarp::os::Network yarp;
if (!yarp::os::Network::checkNetwork(5.0))
{
    yError() << " YARP server not available!";
    return EXIT_FAILURE;
}
```
The problem was solved by defining the Yarp Network **before** the code related to the Resource Finder. Nevertheless, I think it would be beneficial to get an error message to avoid this kind of mistake.

traversaro, 2018-05-03:
@barbalberto

jgvictores, 2018-05-03:
Guess it's all kind of related with PR #1277, similar to issue #1595.

aerydna, 2018-05-24:
As discussed, yarp::os::Network is mandatory for using YARP classes.
We will add documentation soon.

closed 2018-05-24 by aerydna

traversaro, 2018-05-24:
What about keeping this issue open until the documentation is ready?