robotology/yarp #1672: OpenCV troubles

Nicogene, 2018-05-03:
I pulled `opencv` and `yarpdataplayer` and `yarpdatadumper` doesn't compile:
```
/home/ngenesio/yarp/src/yarpdatadumper/main.cpp:418:30: error: use of undeclared identifier 'IplImage'
                int frameW=((IplImage*)itemEnd.obj->getPtr())->width;
                             ^
/home/ngenesio/yarp/src/yarpdatadumper/main.cpp:418:39: error: expected expression
                int frameW=((IplImage*)itemEnd.obj->getPtr())->width;
                                      ^
/home/ngenesio/yarp/src/yarpdatadumper/main.cpp:419:30: error: use of undeclared identifier 'IplImage'
                int frameH=((IplImage*)itemEnd.obj->getPtr())->height;
                             ^
/home/ngenesio/yarp/src/yarpdatadumper/main.cpp:419:39: error: expected expression
                int frameH=((IplImage*)itemEnd.obj->getPtr())->height;
                                      ^
/home/ngenesio/yarp/src/yarpdatadumper/main.cpp:428:52: error: use of undeclared identifier 'CV_FOURCC'
                videoWriter.open(videoFile.c_str(),CV_FOURCC('H','F','Y','U'),
                                                   ^
/home/ngenesio/yarp/src/yarpdatadumper/main.cpp:429:38: error: use of undeclared identifier 'cvSize'
                                 fps,cvSize(frameW,frameH),true);
                                     ^
/home/ngenesio/yarp/src/yarpdatadumper/main.cpp:457:49: error: use of undeclared identifier 'IplImage'
                    cv::Mat img=cv::cvarrToMat((IplImage*)item.obj->getPtr());
                                                ^
/home/ngenesio/yarp/src/yarpdatadumper/main.cpp:457:58: error: expected expression
                    cv::Mat img=cv::cvarrToMat((IplImage*)item.obj->getPtr());
                                                         ^
/home/ngenesio/yarp/src/yarpdatadumper/main.cpp:457:37: error: no member named 'cvarrToMat' in namespace 'cv'
                    cv::Mat img=cv::cvarrToMat((IplImage*)item.obj->getPtr());

```

my `opencv_version` is `4.0.0-pre`, I'm afraid that in opencv 4 they will remove `IplImage` and `cvarrToMat`. We should make `yarp::sig::Image` compatible with `cv::Mat` without taking account of `IplImage`.

drdanz, 2018-05-03:
It could be a good opportunity to get rid of the opencv integration inside `YARP_sig`, and to add a separate library `YARP_opencv`

Nicogene, 2018-05-03:
They moved these classes in different headers or removed the inclusions from the "principal" headers.
Adding : 
```
    #include <opencv2/core/types_c.h>
    #include <opencv2/videoio/videoio_c.h>
```
compiles :+1: 
Probably they did these changes to push people to move to cv:Mat, by the way let's wait the official release of OpenCV 4.0.0 scheduled for July 2018

claudiofantacci, 2018-09-21:
Yes, they are getting rid/transferring C code in dedicated headers.
Anyway, I like the idea proposed by @drdanz and a clean way to integrate our image with opencv which I find somewhat convoluted.

traversaro, 2019-01-29:
@Nicogene can we change the name of the issue to "YARP does not compiled with OpenCV 4" so it is clear what this issue is tracking? Thanks!

traversaro, 2019-01-29:
If the solution is simply add those includes if OpenCV 4 is used, probably it could make sense to add those lines to solve problems such as https://github.com/robotology/yarp/pull/1948?

claudiofantacci, 2019-01-29:
> If the solution is simply add those includes if OpenCV 4 is used, probably it could make sense to add those lines to solve problems such as #1948?

BTW, brew formula [`opencv@3`](https://formulae.brew.sh/formula/opencv@3) is out, we should update our macOS builds accordingly.

traversaro, 2019-01-29:
cc @nunoguedelha

Nicogene, 2019-01-29:
> BTW, brew formula opencv@3 is out, we should update our macOS builds accordingly.

I agree to use `opencv@3`  formula, btw in #1948 there is an issue, the formula is keg-only, then actually in macOs builds `opencv` is not found.
@claudiofantacci if you do the PR to change the formula could you please also fix this issue?

Nicogene, 2019-01-29:
> If the solution is simply add those includes if OpenCV 4 is used

Unfortunately it is not so simple, because what fixes the builds with opencv 4 breaks the builds with opencv 2, and unfortunately both in appveyor and travis(linux) there are old versions of opencv.

We  should use a lot of `#ifdef` or decide to update the CI infrastructure and drop the support for `opencv2`

claudiofantacci, 2019-01-29:
It is keg-only because `opencv` (i.e. `opencv4`) is the most updated formula. If we install `opencv@3` onlyk, we can just force-link it with `brew install --force opencv@3`.

Nicogene, 2019-01-29:
> We are not supporting code with opencv2, right? 

In theory yes, in practice no :sweat_smile: we have `trusty` on travis.
See : https://packages.ubuntu.com/source/trusty/opencv

claudiofantacci, 2019-01-29:
> In theory yes, in practice no 😅 we have `trusty` on travis.
> See : https://packages.ubuntu.com/source/trusty/opencv

Travis now has `xenial` and by my understanding we no longer support `trusty`.
We could just update everything.

nunoguedelha, 2019-01-29:
> It is keg-only because `opencv` (i.e. `opencv4`) is the most updated formula. If we install `opencv@3` only, we can just force-link it with `brew install --force opencv@3`.

I guess you meant `brew link --force opencv@3`...

As a side note, the link to the files in "Cellar" folder is created as `/usr/local/share/OpenCV` and not `/usr/local/lib/cmake/opencv4` as for the normal formula.
Anyway, `CMake` had no trouble finding it.

For me it's building just fine now.

claudiofantacci, 2019-01-29:
> I guess you meant `brew link --force opencv@3`

There is a `--force` option for the `install` command as well that should do it as well. I can be wrong, I can't test it atm.

nunoguedelha, 2019-01-29:
Actually I've tested. Even after uninstalling the `opencv4`, installing the `opencv@3` as you suggested, it would not symlink it. You can always try, better two independent tests than one.

claudiofantacci, 2019-01-29:
Ok, thanks! Good to keep in mind!

lrapetti, 2019-04-03:
> Actually I've tested. Even after uninstalling the `opencv4`, installing the `opencv@3` as you suggested, it would not symlink it. You can always try, better two independent tests than one.

I confirm `brew install --force opencv@3` was not enough. I had to do `brew unlink opencv`, and `brew link --force opencv@3` (without the `unlink` I was not able to link `opencv@3` even if using `--force`).

traversaro, 2019-08-13:
Status update: the default `opencv` vcpkg port now builds opencv4 (see https://github.com/microsoft/vcpkg/pull/5169#issuecomment-520497548). While an `opencv3` port is also still available, I will expect that the availability of opencv4 in vcpkg will mean more user will be impacted by this issue.

drdanz, 2019-08-14:
The main problem here is that debian/ubuntu don't have opencv4 yet (it is currently in debian experimental) therefore we are stuck with supporting opencv3 for a long time, but we can drop opencv2 as long as we drop support for debian stretch (oldstable)
It would be great if we could support opencv4 using only features that are also available in opencv3

drdanz, 2019-11-14:
OpenCV 4.1.3 landed in debian testing. This means that it will also be in the next Ubuntu LTS (20.04).
Therefore supporting OpenCV 4 is now a major issue. @Nicogene if I remember correctly, some time ago you started working on this, did you come to any conclusion?

drdanz, 2019-11-14:
> OpenCV 4.1.3 landed in debian testing. This means that it will also be in the next Ubuntu LTS (20.04).

Just for the records, this means that OpenCV 3 will *no longer be available* on debian/ubuntu

Nicogene, 2019-11-14:
I started working on it some time ago, support both version 3 and 4 was a mess of #ifdef . We have to decide if we want to support only version 4 or both

drdanz, 2019-11-14:
I think we can drop OpenCV 2 since we have the new icub head image with debian buster, and therefore we can drop support for debian stretch, but we will need to support both OpenCV 3 and 4 for at least 2 more years

drdanz, 2019-11-14:
I disabled building YARP with OpenCV 4, since this will only cause compilation issues.

Nicogene, 2019-12-20:
With https://github.com/robotology/yarp/pull/2162 it compiles fine with both OpenCV 3 and 4.

The question is, we want to migrate to the c++ api and leave the old OpenCV 2 api?

cc @pattacini @vtikha @traversaro @drdanz @randaz81 @lornat75 @maggia80

traversaro, 2019-12-20:
> The question is, we want to migrate to the c++ api and leave the old OpenCV 2 api?
> 
> cc @pattacini @vtikha @traversaro @drdanz @randaz81 @lornat75 @maggia80

I think we can merge https://github.com/robotology/yarp/pull/2162, and afterwards we can discuss if there is any advantage migrating to the new API.

pattacini, 2019-12-20:
Agree with @traversaro 
Anyway, my opinion is to get rid of the old C API. When this needs to be done is up to you.

Nicogene, 2020-01-07:
Fixed by #2162

PR #2162 (2020-01-07): Support OpenCV 4
It fixes #1672.

It compiles also with OpenCV 3

Please review code.
closed 2020-01-07 by Nicogene