robotology/yarp #1687: [yarpdataplayer] cannot stream two datapackage from same directory

lpenco, 2018-05-11:
Dear yarp experts, I'm having issues with the yarpdataplayer.
I use Ubuntu 16 and the latest version of yarp from github. When I try to stream more than one data from the same directory, the second datapackage cannot be streamed even though I uncheck the first one. The first one as well never completes the streaming. It gets stuck at 91%. The problem does not appear if streaming just from a folder with one data package

[image]

opening file /home/waldezjr/retargetdata/T-N/jointAngles/info.log
the name of the port is /xsens/JointAngles
opening file /home/waldezjr/retargetdata/T-N/jointAngles/data.log
opening file /home/waldezjr/retargetdata/T-N/pos/info.log
the name of the port is /xsens/PoseQuaternion
opening file /home/waldezjr/retargetdata/T-N/pos/data.log
1524063587,129879
1524735808,463906
the biggest timestamp is: index 1 with value 1524735808,463906
1524063587,129879
1524735808,463906
the smallest timestamp is: index 0 with value 1524063587,129879
The part jointAngles, should last for: 4,039392 with 312 frames
The part pos, should last for: 4,340003 with 540 frames
need to create new port Bottle for jointAngles
creating and opening Bottle port for part jointAngles
yarp: Port /xsens/JointAngles active at tcp://192.168.62.168:10087/
need to create new port Bottle for pos
creating and opening Bottle port for part pos
yarp: Port /xsens/PoseQuaternion active at tcp://192.168.62.168:10056/
checking if port was changed by the user...
Port is the same continue
Port is the same continue
asking the thread to start
initializing the workers...

[here I press the play button with both the datapackages checked]
[INFO]USING OPENCV FOR SENDING THE IMAGES
[INFO]USING OPENCV FOR SENDING THE IMAGES
starting the master thread...
partID: 0 has finished 
asking the threads to stop...
done stopping!
done stopping the thread...
checking if port was changed by the user...

[here I uncheck the first datapackage but the second one is not streamed]
Port is the same continue
Port is the same continue
asking the thread to start
initializing the workers...
[INFO]USING OPENCV FOR SENDING THE IMAGES
[INFO]USING OPENCV FOR SENDING THE IMAGES
starting the master thread...
partID: 0 has finished

lpenco, 2018-05-30:
Here a portion of the two datasets:

[first]
-1 1524063587.129879 0.730584 -0.139337 3.28526 0.323712 -0.0670975 1.4603 0.323712 -0.0670975 1.4603 0.242625 -0.0510968 1.09526 0.0183631 -0.617898 18.6027 -0.167868 -1.13442 -1.63594 0.456093 3.15156 3.22838 79.1869 -24.6671 12.9903 -1.90557 90.5923 6.02212 -3.95161 -0.361651 -2.26777 0.687127 4.3685 12.4477 78.0906 -55.696 -14.196 -5.16369 91.8734 4.05929 0.850878 0.046595 6.53769 4.34718 -8.11723 -6.33144 0.446612 0.493032 6.10611 -4.57979 4.94417 7.49155 -0.00586655 -4.79983e-05 -0.0167586 3.48917 -1.50277 -5.61325 -0.347489 -1.70608 6.86672 -3.88146 1.55174 7.76964 0.0129839 0.000161859 -0.0244552
-1 1524063587.157864 0.649963 -0.160339 3.3018 0.287727 -0.0758841 1.46767 0.287727 -0.0758841 1.46767 0.215614 -0.0576043 1.10078 0.0361238 -0.748631 18.8828 -0.175129 -1.37919 -1.11495 0.461173 3.15229 3.38917 79.1253 -24.5772 12.8616 -1.89332 90.5528 5.97817 -3.97354 -0.337684 -2.1863 0.685308 4.33397 12.5 78.0328 -55.6588 -14.1318 -5.16749 91.9329 4.01649 0.775025 0.175584 6.57403 4.27111 -8.22852 -6.38113 0.47151 0.398737 6.11568 -4.54455 4.99198 7.56554 -0.00106054 -1.33663e-05 7.70395e-05 3.51636 -1.46432 -5.58125 -0.317001 -1.57743 6.90108 -3.90776 1.63444 7.79927 -0.00281564 -3.30839e-05 0.00106377
-1 1524063587.168029 0.649693 -0.16392 3.29942 0.287582 -0.0774702 1.46662 0.287582 -0.0774702 1.46662 0.215502 -0.0587931 1.1 0.0249552 -0.746089 18.8983 -0.194551 -1.37096 -1.08595 0.461249 3.15892 3.39329 79.1415 -24.5817 12.8773 -1.9063 90.5883 5.98485 -3.98292 -0.327902 -2.1805 0.67859 4.33746 12.5057 78.0212 -55.669 -14.151 -5.15832 91.8895 4.00419 0.770211 0.1762 6.57234 4.26856 -8.2484 -6.38777 0.477057 0.401608 6.12279 -4.54404 5.0009 7.57277 -0.00112528 -1.3082e-05 -0.000542577 3.51584 -1.44341 -5.58245 -0.318996 -1.58581 6.90891 -3.90675 1.63506 7.80338 -0.00238686 -2.82221e-05 9.44273e-05
-1 1524063587.172584 0.655 -0.179582 3.29945 0.28983 -0.0844688 1.46665 0.28983 -0.0844688 1.46665 0.217171 -0.0640474 1.10003 0.0355549 -0.764717 18.9035 -0.181129 -1.40841 -1.07641 0.442215 3.10971 3.39338 79.1242 -23.7725 13.6747 -1.92116 90.5999 5.98589 -4.00324 -0.34534 -2.19431 0.68788 4.30616 12.4887 78.0014 -55.6464 -14.131 -5.24016 91.8682 4.05474 0.665782 0.0537901 6.50619 4.27872 -8.21236 -6.40074 0.469386 0.412081 6.11818 -4.54344 4.83725 7.5722 -0.0117526 -0.000122256 -0.0144533 3.51209 -1.50966 -5.59401 -0.322122 -1.5559 6.90388 -3.90495 1.70973 7.792 0.0145157 0.000170191 -0.0180165
-1 1524063587.184652 0.654264 -0.18197 3.29662 0.289486 -0.0855205 1.4654 0.289486 -0.0855205 1.4654 0.216911 -0.0648348 1.09909 0.0307146 -0.765448 18.9156 -0.190077 -1.40819 -1.05384 0.436454 3.10664 3.39497 79.1305 -23.7287 13.717 -1.92554 90.5947 5.99781 -3.99694 -0.348004 -2.19689 0.686088 4.30578 12.4839 77.993 -55.6366 -14.1083 -5.2358 91.8772 4.04697 0.676441 0.0508894 6.48061 4.27851 -8.21669 -6.40817 0.470489 0.40832 6.12089 -4.54362 4.84201 7.57635 -0.00647662 -6.63324e-05 -0.00835749 3.50823 -1.50401 -5.59738 -0.317442 -1.55541 6.90592 -3.9059 1.71233 7.79485 0.00699888 8.14758e-05 -0.00876154
-1 1524063587.193212 0.652429 -0.183278 3.29412 0.288663 -0.0860854 1.46429 0.288663 -0.0860854 1.46429 0.216292 -0.065256 1.09825 0.0266858 -0.763972 18.9251 -0.196861 -1.40421 -1.03605 0.433728 3.10678 3.39277 79.1314 -23.7214 13.7089 -1.92042 90.5621 6.00486 -3.97724 -0.354444 -2.20166 0.683728 4.30826 12.478 77.9894 -55.6197 -14.0822 -5.23202 91.902 4.04259 0.690031 0.0573038 6.4643 4.27672 -8.21638 -6.41319 0.472888 0.395008 6.12576 -4.54124 4.84485 7.58054 -0.00465464 -5.01315e-05 -0.00446263 3.50278 -1.49463 -5.60067 -0.313386 -1.55001 6.90814 -3.90429 1.71221 7.79527 0.00231905 2.66805e-05 -0.00506126
-1 1524063587.200553 0.651732 -0.184096 3.29306 0.288347 -0.0864426 1.46382 0.288347 -0.0864426 1.46382 0.216055 -0.0655229 1.0979 0.025907 -0.763331 18.928 -0.198048 -1.40279 -1.03083 0.432317 3.10714 3.39053 79.1305 -23.7108 13.7149 -1.91742 90.5516 6.00536 -3.97147 -0.355634 -2.20322 0.683482 4.3092 12.4748 77.9874 -55.6235 -14.085 -5.2328 91.9089 4.04476 0.692452 0.0603624 6.46364 4.27592 -8.21325 -6.41517 0.473894 0.386638 6.12715 -4.54151 4.84602 7.58162 -0.00387621 -4.14949e-05 -0.00385545 3.50246 -1.48905 -5.60068 -0.313362 -1.54911 6.91022 -3.90405 1.71084 7.79608 0.00147472 1.68716e-05 -0.00403437
-1 1524063587.228765 0.646305 -0.19943 3.27641 0.285834 -0.0931959 1.45643 0.285834 -0.0931959 1.45643 0.214155 -0.0705784 1.09237 0.0204099 -0.757437 18.9466 -0.206 -1.39028 -0.996263 0.430791 3.11951 3.36757 79.1147 -23.7202 13.7288 -1.91716 90.5606 6.00524 -3.97318 -0.345888 -2.18381 0.684093 4.29547 12.4422 77.9809 -55.5897 -14.0808 -5.24793 91.8983 4.06807 0.678678 0.0807436 6.49529 4.26205 -8.20597 -6.42892 0.483621 0.346591 6.12677 -4.54052 4.84833 7.58609 -0.00102167 -1.04825e-05 -0.00126794 3.505 -1.47764 -5.59474 -0.31215 -1.52815 6.92827 -3.90512 1.71869 7.8057 -0.000773427 -8.97541e-06 0.000339354
-1 1524063587.242449 0.64454 -0.208193 3.26479 0.284993 -0.0970615 1.45128 0.284993 -0.0970615 1.45128 0.213515 -0.0734732 1.0885 0.017591 -0.75696 18.9551 -0.210908 -1.38851 -0.980485 0.431253 3.13007 3.355 79.1086 -23.7935 13.6804 -1.92794 90.5838 6.00494 -3.99132 -0.340421 -2.15885 0.684798 4.28043 12.4263 77.9799 -55.5224 -14.0376 -5.25727 91.8738 4.08102 0.680672 0.0850157 6.50675 4.26131 -8.22012 -6.43595 0.480891 0.337601 6.12322 -4.54228 4.85446 7.5875 0.000579052 8.21748e-06 -0.000671358 3.51237 -1.47742 -5.58779 -0.314715 -1.50906 6.9375 [... cut ...] -0.271327 1.02839 57.2639 -1.90899 -0.00131644 -0.0692227 0.138795 -0.301399 0.577851 57.1751 -2.11902 3.05323 0.0552848 0.0911257 -0.334558 0.131898 57.2275 0.134189 -0.706722 2.70188 0.261021 -0.318083 0.0219844 57.2275 0.132378 -0.706897 2.70188 0.114506 -0.100972 1.02989 57.0601 2.01527 -0.00226891 4.78397 0.111902 -0.0693338 0.579472 56.8907 1.615 3.37104 5.68162 0.0567782 -0.0494298 0.133554 57.0966 0.0562043 -0.672366 4.7251 0.225005 -0.0213913 0.0234152 57.0966 0.0552444 -0.672064 4.7251

vtikha, 2018-06-05:
@lpenco how did you collect this dataset? The `yarpdataplayer` tries to load a dataset and synchronise its contents with respect to the various timestamps. Here you have two timestamps well different from each other **`1524063587,129879`** and **`1524735808,463906`** and that is why the player has difficulties in running this. Please attach the files instead of copying their data in the issue and I will try it locally.

lpenco, 2018-06-11:
Ok I solved with the synchronization! Before I was collecting the data one at a time
Thanks a lot

closed 2018-06-11 by lpenco