robotology/yarp #1693: Provide a wrapper device to write data read from the Multiple Analog Sensors Interfaces to a vector

traversaro, 2018-05-21:
Given that several YARP workflows (such as using the yarpscope) rely on reading data from a port in which data is published in a port as `yarp::sig::Vector`, it would be great to have a configurable wrapper YARP device that attaches to a device exposing [Multiple Analog Sensor Interfaces](http://yarp.it/devel/group__dev__iface__multiple__analog.html) and publishes them as a vector. 

Note that this wrapper  device would not have any associated client, as the network transparency would be always implemented with the [`yarp::dev::MultipleAnalogSensorsServer`](http://yarp.it/devel/classyarp_1_1dev_1_1MultipleAnalogSensorsServer.html)
and the [`yarp::dev::MultipleAnalogSensorsClient`](http://yarp.it/devel/classyarp_1_1dev_1_1MultipleAnalogSensorsClient.html) 

**Describe alternatives you've considered**
A possible alternative is to have a YARP Module that opens a `multipleanalogsensorsclient`and publish the necessary info on a port, but implementing this logic in a device gives us more flexibility, as the device could be added directly in the `yarprobointerface` to reduce delays. 

cc @hu-yue @fjandrad

barbalberto, 2018-05-21:
What about a `portmonitor` plugin that strips away the unnecessary header/data and formats the output as a vector?

DanielePucci, 2018-05-21:
CC
@diegoferigo

jgvictores, 2020-01-15:
Hi! Here's 1 solution from our repos, and 2 questions regarding a possible future solution contained within this repo. :smile: 

**Solution**: We identified this issue at https://github.com/roboticslab-uc3m/teo-configuration-files/issues/27, unaware of this duplicate. As a result, we developed a [BottleExtract](https://github.com/roboticslab-uc3m/yarp-devices/blob/e3b9d8e0874c75d26a1b0e05d9d63bcf805aff76/libraries/YarpPlugins/PortMonitorPlugins/cpp/bottle_extract_portmonitor/BottleExtract.cpp) YARP portmonitor, which is functional and can be used with a format such as `tcp+recv.portmonitor+type.dll+file.BottleExtract+index.5+subindex.0+subsubindex.0` ([example for plotting FT on `yarpscope`](https://github.com/roboticslab-uc3m/teo-configuration-files/blob/7c3a88329cdde9dac196e692368b445fa83a2545/share/applications/teoSimTools.xml#L17-L40)).

**Questions**: Some questions I'd ask if we were interested in providing a more formal version in this repository:
1. The fact that `recv.portmonitor` (can be casted into a `Bottle`) works but `send.portmonitor` (cannot be casted) feels kind or risky/hacky. Would this be acceptable, or would we really have to make a MAS-specific device with, e.g. access to `SensorStreamingData` header created by Thrift IDL?
2. There is some `prop.val+prop+val`->`(prop val) (prop val)` logic at [BottleExtract](https://github.com/roboticslab-uc3m/yarp-devices/blob/e3b9d8e0874c75d26a1b0e05d9d63bcf805aff76/libraries/YarpPlugins/PortMonitorPlugins/cpp/bottle_extract_portmonitor/BottleExtract.cpp#L22-L33) that is a kind of re-implementation of what is already done at YARP's [Protocol.cpp#L149-L175](https://github.com/robotology/yarp/blob/d62a3903a05e96005de4133ab8553ffb214c0dd0/src/libYARP_os/src/yarp/os/impl/Protocol.cpp#L149-L175). The parsing logic is accessed by `PortMonitor` [here](https://github.com/robotology/yarp/blob/d62a3903a05e96005de4133ab8553ffb214c0dd0/src/carriers/portmonitor_carrier/PortMonitor.cpp#L37) but not passed on to the loaded `.dll` (obtains a good `Property` [here](https://github.com/robotology/yarp/blob/d62a3903a05e96005de4133ab8553ffb214c0dd0/src/carriers/portmonitor_carrier/PortMonitor.cpp#L36-L45) but instead of passing it directly it only passes parts of it on a new `Property` [here](https://github.com/robotology/yarp/blob/d62a3903a05e96005de4133ab8553ffb214c0dd0/src/carriers/portmonitor_carrier/PortMonitor.cpp#L82-L95), where the `carrier` string is again unparsed). Knowing this would be helpful to not duplicate the code for the `create()` any plugin of `category portmonitor` (e.g. I'd like to avoid [hard-coded initial values in `depth_image_portmonitor`](https://github.com/robotology/yarp/blob/d62a3903a05e96005de4133ab8553ffb214c0dd0/src/carriers/depth_image_portmonitor/DepthImage.cpp#L23-L24) in the future), would it be okay to touch the `PortMonitor` even if it is to pass, say, both the unparsed `carrier` string **as well as** the parsed version e.g. as a nested `Property`?

traversaro, 2020-01-15:
Thanks @jgvictores for the interesting links. I am not a PortMonitor expert so I am not sure how to answers your questions. However, I think there are a few users who could be interested in your comment: @Yeshasvitvs @lrapetti @diegoferigo @prashanthr05 @HosameldinMohamed .

traversaro, 2021-06-15:
Related PR from @randaz81 : https://github.com/robotology/yarp/pull/2610 .