robotology/yarp #1701: What does getTorqueRange() return?

lrapetti, 2018-05-24:
Working at https://github.com/robotology/gazebo-yarp-plugins/pull/355 I end up not understanding what _getTorqueRange()_ should return.

In the documentation it is defined as:
> full scale of the torque sensor of a given joint

However it is not directly related to a torque sensor (that is not presented), while it is a method of `ITorqueControl`. So it may be interpreted as the maximum torque that can be asked from the controller (different from the maximum torque that the joint can reach).

traversaro, 2018-05-24:
cc @randaz81 ?

randaz81, 2018-05-24:
I'm confused by many PRs stating different things. Please let's discuss f2f which changes you want to apply to yarp interfaces.

traversaro, 2018-05-25:
I discussed f2f with @randaz81 : the `getTorqueRange()` is used to pass to the user a reasonable range of values in which the `RefTorques` could be set to the low level joint torque control, so probably in gazebo `m_maxTorques` is not the right quantity to return. Closing the issue as the doubt have been solved.