robotology/yarp #1705: In the devel branch launching `serial` device with `serialport` subdevice results in the `serial` device launched twice

traversaro, 2018-05-25:
**Describe the bug**
Launching `serial` device with `serialport` subdevice results in the `serial` device launched twice

**To Reproduce**
Steps to reproduce the behavior:
~~~
yarpdev --device serial --subdevice serialport  --comport /dev/ttyACM0
~~~ 
where `/dev/ttyACM0` is a valid serial device

**Expected behavior**
The `serial` device should be launched only once. 
This is what happens for example if you launch `yarpdev --device inertial --subdevice fakeIMU` .

**Configuration (please complete the following information):**
 - OS:
 - yarp version: 2.3.72 and 2.3.73
-  compiler:GCC 

**Additional context**
Behaviour on the `master` branch: 
~~~bash
straversaro@iiticublap103:~/src/robotology-superbuild/robotology/YARP$ yarpdev --device serial --subdevice serialport  --comport /dev/ttyACM0
[DEBUG]Subdevice serialport
[INFO]Starting Serial Port in /dev/ttyACM0 
[INFO]created device <serialport>. See C++ class yarp::dev::SerialDeviceDriver for documentation.
yarp: Port /serial/in active at tcp://10.255.111.14:10002/
yarp: Port /serial/out active at tcp://10.255.111.14:10003/
[INFO]created wrapper <serial>. See C++ class yarp::dev::ServerSerial for documentation.
[INFO]Server Serial starting
yarp: Port /serial/quit active at tcp://10.255.111.14:10004/
~~~

Behaviour on the `devel` branch:
~~~bash
straversaro@iiticublap103:~/src/robotology-superbuild/robotology/YARP$ yarpdev --device serial --subdevice serialport  --comport /dev/ttyACM0
[DEBUG]Subdevice serialport
[DEBUG]Subdevice serialport
[INFO]Starting Serial Port in /dev/ttyACM0 
[INFO]created device <serialport>. See C++ class yarp::dev::SerialDeviceDriver for documentation.
yarp: Port /serial/in active at tcp://10.255.111.14:10002/
yarp: Port /serial/out active at tcp://10.255.111.14:10003/
[INFO]Server Serial starting
[INFO]created device <serial>. See C++ class yarp::dev::ServerSerial for documentation.
yarp: Port /serial/in failed to activate at tcp://10.255.111.14:10002/ (address conflict)
yarp: Port /serial/out failed to activate at tcp://10.255.111.14:10003/ (address conflict)
[INFO]created device <serial>. See C++ class yarp::dev::ServerSerial for documentation.
[INFO]Server Serial starting
yarp: Port /serial/quit active at tcp://10.255.111.14:10004/
[INFO]device active in background...
~~~

traversaro, 2018-05-25:
Note: this affects all the iCub robots when launching the [`faceExpressions`](https://github.com/robotology/icub-main/tree/master/app/faceExpressions) device.

If you try to replicate this issue, remember to delete every time the build/install directory of YARP, as the different configuration of `yarp/share/plugins/serial.ini` and `yarp/share/plugins/serialport.ini` will confuse the YARP plugin framework a lot.

traversaro, 2018-05-25:
I think this was introduced in https://github.com/robotology/yarp/commit/16d8ad30799c47fe1eb2a0486d32bcf508558c3d , but I did not checked this. 

@drdanz @barbalberto any idea?

traversaro, 2018-05-25:
cc @hu-yue @GiulioRomualdi

traversaro, 2018-05-25:
Related issue: https://github.com/robotology/yarp/issues/1547 .

Nicogene, 2018-05-25:
Try to modify ServerSerial.cpp in order to have 

`yarp::dev::DriverCreator *createServerSerial() {
    return new yarp::dev::DriverCreatorOf<yarp::dev::ServerSerial>("serial",
                                                                   "serial",
                                                                   "yarp::dev::ServerSerial");`
Putting the itself as wrapper makes it a network wrapper.
Not sure this fix it, but it is the only difference from other wrappers.

traversaro, 2018-05-25:
That's great @Nicogene , this fixes it.

traversaro, 2018-05-25:
Fixed by https://github.com/robotology/yarp/pull/1706 .

closed 2018-05-25 by drdanz