robotology/yarp #1720: opencv grabber

mike-talmage, 2018-06-01:
opencv_grabber compiles fine but when i try to use it it says

yarpdev: symbol lookup error: /usr/local/lib/yarp/yarp_opencv.so: undefined symbol: _ZN2cv12VideoCaptureC1Ev

tried it several times every time it does this

traversaro, 2018-06-01:
On which OS are you experiencing this? If Debian/Ubuntu, which version?

mike-talmage, 2018-06-01:
ubuntu 16.04

Nicogene, 2018-06-01:
Did you forget to do `make install` after compiling?

mike-talmage, 2018-06-01:
no every thing works did make install and it went perfect

mike-talmage, 2018-06-01:
no errors or warnings

mike-talmage, 2018-06-01:
using yarp 2.3.70

mike-talmage, 2018-06-01:
if I have to go up its going to be hard my iot only compiles this version

what can I do

PeterBowman, 2018-06-13:
What is the output of `ldd /usr/local/lib/yarp/yarp_opencv.so | grep opencv`?

drdanz, 2018-06-14:
This is something related to your configuration, YARP does not compile if one of the symbols is missing.
Please try again with a clean build...

Nicogene, 2018-06-20:
@mike-talmage any news?
The problem is solved? Shall we close it?

Nicogene, 2018-07-11:
Closing for now, @mike-talmage  feel free to reopen it.

closed 2018-07-11 by Nicogene