robotology/yarp #1722: Macro error in Storable

diegoferigo, 2018-06-04:
**Describe the bug**

Building the devel branch fails with the following error:

```
[  3%] Building CXX object src/libYARP_OS/CMakeFiles/YARP_OS.dir/src/Storable.cpp.o
/tmp/robotology-superbuild/robotology/YARP/src/libYARP_OS/src/Storable.cpp:77:24: error: use of undeclared identifier 'FAILED_TO_RUN'
    char buf[16 + 15 + FAILED_TO_RUN];
                       ^
/tmp/robotology-superbuild/robotology/YARP/src/libYARP_OS/src/Storable.cpp:78:34: error: use of undeclared identifier 'FAILED_TO_RUN'
    std::snprintf(buf, 16 + 15 + FAILED_TO_RUN, "%.*g", 15, x);
```

**Configuration (please complete the following information):**
 - OS: ubuntu
 - yarp version: commit 50c6677bf126ea88259eb9f4d77334bc45e2f464
-  compiler: clang6

**Additional context**

The error is generated here:

https://github.com/robotology/yarp/blob/50c6677bf126ea88259eb9f4d77334bc45e2f464/src/libYARP_OS/src/Storable.cpp#L77-L79

Probably due to a failing substitution of:

https://github.com/robotology/yarp/blob/50c6677bf126ea88259eb9f4d77334bc45e2f464/src/libYARP_OS/src/Storable.cpp#L67

in the following CMake file:

https://github.com/robotology/yarp/blob/245690a2bb2b371a81efe0da2f9dbbd9f34e53b2/src/libYARP_conf/CMakeLists.txt#L57-L61

 cc @drdanz @Nicogene  @traversaro

drdanz, 2018-06-14:
I'd say that the problem is in the `try_run` here:

https://github.com/robotology/yarp/blob/245690a2bb2b371a81efe0da2f9dbbd9f34e53b2/cmake/YarpSystemCheck.cmake#L144-L167

Did it happen once, or is this something reproducible?

diegoferigo, 2018-06-15:
In the pre-release days this was fully reproducible and it blocked me for like two days (I downgraded temporarily). The tagged `v3.0.0` instead works flawlessly, I don't know what did change but the problem didn't occur again.

Nicogene, 2018-06-15:
Cool closing for now, if it happens again please reopen it!

closed 2018-06-15 by Nicogene