robotology/yarp #1723: InstallBasicPackageFiles problem

diegoferigo, 2018-06-04:
**Describe the bug**

Configuring a fresh build folder with the latest yarp devel commit, I get the following configuring error:

```
CMake Error at /iit/local/share/YCM/modules/InstallBasicPackageFiles.cmake:247 (message):
  TARGETS_PROPERTY or TARGET_PROPERTIES argument is required
Call Stack (most recent call first):
  cmake/YarpInstallBasicPackageFiles.cmake:48 (install_basic_package_files)
  src/libYARP_conf/CMakeLists.txt:89 (yarp_install_basic_package_files)
```

It is likely related to the latest changes in YCM. @drdanz 

**Configuration (please complete the following information):**
 - OS: ubuntu 16.04
 - yarp version: 50c6677bf126ea88259eb9f4d77334bc45e2f464
-  compiler: clang6
- cmake 3.5.1 and 3.9.6

diegoferigo, 2018-06-04:
Source of this regression: https://github.com/robotology/yarp/commit/2a6bb4cb23051e7a9c044ff31aa2a98d4dcd2a25.

lrapetti, 2018-06-04:
Workaround for having a working YARP+ICUB close to devel: 
* Use YARP at commit af5236a3df831d89a20bbe675dbe510b64250d73 
* Use ICUB at commit 8ee55fa75293b9900efa7b23eebcc2a61996ff75

drdanz, 2018-06-04:
Try updating YCM to the latest devel...

drdanz, 2018-06-05:
Fixed by #1728

closed 2018-06-05 by drdanz