robotology/yarp #1724: Add ClockServer thrift file

diegoferigo, 2018-06-04:
**Is your feature request related to a problem? Please describe.**

This is a reminder for the discussion about storing `clock_rpc.thrift` into Yarp started at https://github.com/robotology/gazebo-yarp-plugins/pull/345.

cc @triccyx @traversaro @drdanz

drdanz, 2021-03-11:
I don't remember exactly this discussion, but I think it was something related to gazebo-yarp-plugins, not YARP.
I'm closing this, please reopen if this is not the case.

traversaro, 2021-03-11:
> 
> 
> I don't remember exactly this discussion, but I think it was something related to gazebo-yarp-plugins, not YARP.
> I'm closing this, please reopen if this is not the case.

The point was that gazebo-yarp-plugins is exposing a service to start/stop/pause the simulation, and that service is completely simulation independent, so it could make sense to move it in YARP so that it can be reused by multiple simulators without the need to depend on gazebo-yarp-plugins (that would be odd) or to copy and paste it in every repo that uses it.

drdanz, 2021-03-11:
Oh ok, I see... I'll reopen it, even though I still don't know where this could be in YARP