robotology/yarp #1727: YARP 2.3.73.5 (current devel branch) broken with CMake 3.5

traversaro, 2018-06-04:
Originally reported by @PeterBowman in 

> https://github.com/robotology/yarp/blob/57ed8688762e5fbaf7dfa5d202a0427e314f2ee3/cmake/template/YARPConfig.cmake.in#L142-L144
>
>This is breaking builds in YARP downstreams on Xenial (CMake 3.5.1). Support for additional parameters to `find_dependency()` was added at CMake 3.9 (which is used by Travis, hence no CI build errors).
>
> Docs:
> * https://cmake.org/cmake/help/v3.5/module/CMakeFindDependencyMacro.html
> * https://cmake.org/cmake/help/v3.9/module/CMakeFindDependencyMacro.html
> 
> cc @drdanz

PeterBowman, 2018-06-04:
Right now we have:

https://github.com/robotology/yarp/blob/57ed8688762e5fbaf7dfa5d202a0427e314f2ee3/cmake/template/YARPConfig.cmake.in#L139-L147

This should work at the upmost level (*YARPConfig.cmake*):

```cmake
# Find all requested components
set(YARP_LIBRARIES YARP::YARP_init)
foreach(_module ${YARP_FIND_COMPONENTS})
  include("${YARP_CMAKECONFIG_DIR}/YARP_${_module}/YARP_${_module}Config.cmake"
          RESULT_VARIABLE _have_yarp_${_module})
  if(NOT _have_yarp_${_module})
    set(YARP_FOUND FALSE)
    set(YARP_NOTFOUND_MESSAGE "Missing component: YARP::YARP_${_module}")
  else()
    list(APPEND YARP_LIBRARIES YARP::YARP_${_module})
  endif()
  unset(_have_yarp_${_module})
endforeach()
```

I just followed the [CMake manual](https://cmake.org/cmake/help/v3.7/manual/cmake-packages.7.html). However, specific components (e.g. *YARP_devConfig.cmake*) will still fail, I presume something at *InstallBasicPackageFiles.cmake* have to be changed, too.

PeterBowman, 2018-06-04:
> I presume something at InstallBasicPackageFiles.cmake have to be changed, too.

Correction: [YarpInstallBasicPackageFiles.cmake](https://github.com/robotology/yarp/blob/eb3005e6e51d1dbfd7257772dc4c03370c1cbe4b/cmake/YarpInstallBasicPackageFiles.cmake#L19).

PeterBowman, 2018-06-04:
It looks like YARP components are being treated as separate **packages**, that is, one could do:

```cmake 
find_package(YARP_OS)
find_package(YARP_dev)
# etc.
```

Not sure if this was intended, seems a bit overkill. All export sets could be installed into the same system directory (e.g. `/usr/local/lib/cmake/YARP`), now they are scattered around as quite complete CMake packages (config file + targets file + version file) in sibling folders.

drdanz, 2018-06-05:
@PeterBowman This was intended, it follows the same convention used by Qt5... It simplifies creating separate binary packages. Also it allows one to install new "YARP" packages without actually installing stuff in YARP directory. I'm not saying this is the right decision (perhaps it is not, I wasn't 100% sure about this), but I had to choose between the two options, and after a few tested I was in favour of separate packages, and for now I'm sticking with this decision...

drdanz, 2018-06-06:
Fixed by #1731

PR #1731 (2018-06-06): InstallBasicPackageFiles: Support for CMake < 3.9
This should fix #1727, but I don't have a system to test in this moment.

Can anyone with CMake 3.5 test this, please? @traversaro @PeterBowman @pattacini
closed 2018-06-06 by drdanz