robotology/yarp #1729: devel branch: not installing new devices

lrapetti, 2018-06-04:
I have a problem in the installation of a user-defined `yarpdev` after moving from `master` to `devel` branch.

I have updated the device accordingly to the changes on YARP, and it is compiled and installed without errors. But when I try to use `yarpdev` I get the error:
```
error while opening /home/lrapetti/element_torque-control-via-current/device/motorCurrentControl/build/install/lib/yarp/motorCurrentControl.so:
  /home/lrapetti/element_torque-control-via-current/device/motorCurrentControl/build/install/lib/yarp/motorCurrentControl.so: undefined symbol: motorCurrentControl
Cannot load plugin from shared library (motorCurrentControl)
(motorCurrentControl: cannot open shared object file: No such file or directory)
yarpdev: ***ERROR*** could not find device <motorCurrentControl>
[ERROR]yarpdev: ***ERROR*** device not available.
```

traversaro, 2018-06-05:
Can you copy&paste the CMake code used to configure and compiled the YARP device? Thanks!

lrapetti, 2018-06-05:
```

project(motorCurrentControlDevice)

find_package(YARP REQUIRED)
# ICUB is required for skinDynLib
find_package(ICUB REQUIRED)
find_package(Eigen3 REQUIRED)

list(APPEND CMAKE_MODULE_PATH "${YARP_MODULE_PATH}")
include(YarpInstallationHelpers)

option(BUILD_SHARED_LIBS "Build shared libs?" ON)

yarp_configure_plugins_installation(motorCurrentControlDevice)

yarp_prepare_plugin(passThroughControlBoard CATEGORY device
                                            TYPE yarp::dev::PassThroughControlBoard
                                            INCLUDE PassThroughControlBoard.h
                                            DEFAULT ON
                                            ADVANCED
                                            EXTRA_CONFIG WRAPPER=<controlboardwrapper>)
yarp_prepare_plugin(motorCurrentControl CATEGORY device
                                       TYPE yarp::dev::motorCurrentControl
                                       INCLUDE motorCurrentControl.h
                                       DEFAULT ON
                                       ADVANCED
                                       EXTRA_CONFIG WRAPPER=<controlboardwrapper>)

set(ENABLE_motorCurrentControl TRUE)

if(ENABLE_motorCurrentControl)
    include_directories(${CMAKE_CURRENT_SOURCE_DIR})

    include_directories(SYSTEM
                        ${EIGEN3_INCLUDE_DIR}
                        ${YARP_INCLUDE_DIRS}
    			${skinDynLib_INCLUDE_DIRS})

    yarp_add_plugin(motorCurrentControl motorCurrentControl.h motorCurrentControl.cpp PassThroughControlBoard.h  PassThroughControlBoard.cpp)
    target_link_libraries(motorCurrentControl ${YARP_LIBRARIES})

    yarp_add_plugin(passThroughControlBoard PassThroughControlBoard.h PassThroughControlBoard.cpp)
    target_link_libraries(passThroughControlBoard ${YARP_LIBRARIES})

    if(MSVC)
       add_definitions(-D_USE_MATH_DEFINES)
    endif()

    yarp_install(TARGETS passThroughControlBoard
                 COMPONENT runtime
                 LIBRARY DESTINATION ${YARP_DYNAMIC_PLUGINS_INSTALL_DIR}
                 ARCHIVE DESTINATION ${YARP_DYNAMIC_PLUGINS_INSTALL_DIR})
                 
    yarp_install(TARGETS motorCurrentControl
                 COMPONENT runtime
                 LIBRARY DESTINATION ${YARP_DYNAMIC_PLUGINS_INSTALL_DIR}
                 ARCHIVE DESTINATION ${YARP_DYNAMIC_PLUGINS_INSTALL_DIR})

    add_subdirectory(app)
    
    yarp_install(FILES motorCurrentControl.ini DESTINATION ${YARP_PLUGIN_MANIFESTS_INSTALL_DIR})
endif()

```

traversaro, 2018-06-06:
We debugged the problem: apparently if a [`cmake_minimum_required`](https://cmake.org/cmake/help/v3.5/command/cmake_minimum_required.html) command is missing in the top of the project, `yarp_prepare_plugin` and `yarp_add_plugin` are not working properly due to the [`CMP0012`](https://cmake.org/cmake/help/v3.0/policy/CMP0012.html) not being set to `NEW`. 

We can close as it is, or we can modify `YARPConfig` to have a check on `CMAKE_MINIMUM_REQUIRED_VERSION` to make sure not to have this problems in the future. 

cc @drdanz What do you think?

lrapetti, 2018-06-06:
The cmake minum version was missing in the CMakeList, so it was not compiling all the files.

Solved adding `cmake_minimum_required(VERSION 3.5)`.

lrapetti, 2018-06-06:
The Cmake version required was not specified, so it was not compiling some files.

The problem for me is solved by adding `cmake_minimum_required(VERSION 3.5)` in the `CMakeLists`.

drdanz, 2018-06-14:
> We can close as it is, or we can modify `YARPConfig` to have a check on `CMAKE_MINIMUM_REQUIRED_VERSION` to make sure not to have this problems in the future. 
>
>cc @drdanz What do you think? 

Not sure about this... I think CMake is supposed to print several warning here: 1) when not using `cmake_minimum_required` and 2) in the incriminated line when `CMP0012` is not set. Am I wrong here?

traversaro, 2020-11-18:
> Not sure about this... I think CMake is supposed to print several warning here: 1) when not using `cmake_minimum_required` and 2) in the incriminated line when `CMP0012` is not set. Am I wrong here?

I think that is true, so probably we can just close the issue.

closed 2020-11-18 by drdanz