robotology/yarp #1735: In devel branch YARP::YARP_rtf and YARP::YARP_manager imported targets if only find_package(YARP) is called, even if they are available

traversaro, 2018-06-07:
**Describe the bug**
In devel branch YARP::YARP_rtf and YARP::YARP_manager imported targets if only find_package(YARP) is called, even if they are available .

**To Reproduce**
Just call `find_package(YARP REQUIRED)` and notice that `YARP::YARP_rtf` and `YARP::YARP_manager` are not defined. See for example https://github.com/robotology/icub-tests/blob/devel/CMakeLists.txt#L15 . 

**Expected behavior**
`YARP::YARP_rtf` and `YARP::YARP_manager` should be defined, as it happened in YARP 2.3.72 . 

**Configuration (please complete the following information):**
 - OS: Ubuntu 16.04 
 - yarp version: e0b0607a430ae8d24ef50ae36871205cd4636ac7 
-  cmake: 3.5

drdanz, 2018-06-07:
I don't agree on this, `YARP_rtf` and `YARP_manager` are very specific libraries, if you need them you should enable them explicitly with the relative component...

traversaro, 2018-06-07:
If we want to have a breaking change I am ok with this, but at least it should be documented in the release notes.

drdanz, 2018-06-07:
I just realized I lost somewhere the release notes about the `COMPONENTS` PR, I'm 100% sure I wrote something, but maybe in the thousands of rebases I lost the relative commit :sweat_smile:

traversaro, 2018-06-07:
Things  that need to be updated after this change: 
* https://github.com/robotology/icub-tests
* https://github.com/robotology-playground/ihmc-ors-yarp @gabrielenava  
* All the repos in https://github.com/vvv-school @pattacini @vtikha @diegoferigo @arrenglover @GiuliaP (please add anyone else) 
* Some IIT internal repos @jeljaik @triccyx

drdanz, 2018-06-07:
Done: d4b4aeb31f2fd8645298ae21c2971b2505dd98f8

traversaro, 2018-06-07:
Thanks. The components available are documented somewhere? I had to search in the CMake conf files if the name of the components was `RTF` or `rtf` .

drdanz, 2018-06-07:
@traversaro Nope, also because I'm not 100% if all the components will be stable (i.e., at some point I would like to merge `run`+`companion`+`serversql` and perhaps `manager`+`logger`+`profiler`).

These are the components available at the moment:

**Stable**

* `conf`
* `dev`
* `eigen`
* `gsl`
* `math`
* `OS` (even though this could be changed to `os` in YARP 4 :wink: )
* `pcl`
* `rosmsg`
* `rosmsg_actionlib_msgs`
* `rosmsg_diagnostic_msgs`
* `rosmsg_geometry_msgs`
* `rosmsg_native`
* `rosmsg_nav_msgs`
* `rosmsg_sensor_msgs`
* `rosmsg_shape_msgs`
* `rosmsg_std_msgs`
* `rosmsg_stereo_msgs`
* `rosmsg_tf`
* `rosmsg_tf2_msgs`
* `rosmsg_trajectory_msgs`
* `rosmsg_visualization_msgs`
* `rtf`
* `sig`

**Unstable**

* `companion`
* `logger`
* `manager`
* `name`
* `profiler`
* `run`
* `serversql`
* `wire_rep_utils`

closed 2018-06-20 by drdanz