robotology/yarp #1736: YARP_managerConfig.cmake requires FindTinyXML.cmake to be in CMAKE_MODULE_PATH, but if YARP is configured with YCM it is not

traversaro, 2018-06-07:
**Describe the bug**
If you have `find_package(YARP COMPONENTS manager REQUIRED)` using a YARP installed with YCM, you get the following error: 
~~~
CMake Error at /usr/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:65 (find_package):
  By not providing "FindTinyXML.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "TinyXML", but
  CMake did not find one.

  Could not find a package configuration file provided by "TinyXML" with any
  of the following names:

    TinyXMLConfig.cmake
    tinyxml-config.cmake

  Add the installation prefix of "TinyXML" to CMAKE_PREFIX_PATH or set
  "TinyXML_DIR" to a directory containing one of the above files.  If
  "TinyXML" provides a separate development package or SDK, be sure it has
  been installed.
~~~

**To Reproduce**
Configure a downstream project with  `find_package(YARP COMPONENTS manager REQUIRED)` in its CMakeLists.txt , do not have `find_package(YCM)` in the same downstream project and make sure that YARP was configured/installed using an external YCM 

**Expected behavior**
`find_package(YARP COMPONENTS manager REQUIRED)` should work fine. 

**Configuration (please complete the following information):**
 - OS: Ubuntu 16.04 
 - yarp version: e0b0607a430ae8d24ef50ae36871205cd4636ac7 
-  cmake: 3.5 

**Proposed solution** 
If `YARP` is configured using an external YCM (and so it is not installing YCM CMake modules) a `find_dependency(YCM)` should be added in `YARPConfig.cmake` .

traversaro, 2018-06-07:
@drdanz If you agree on the proposed solution, I can proceed in providing a PR.

traversaro, 2018-06-07:
Apparently I was wrong, and the same problem occurs if YARP is generated without YCM: https://travis-ci.org/robotology/icub-tests/jobs/389236808#L5408 .

PeterBowman, 2018-06-10:
I see this error when I uninstall YCM, though. Could you have installed YCM **master** by chance, and are you sure that YARP is in fact using an external YCM? YARP requires YCM 0.7.0 since quite recently (https://github.com/robotology/yarp/commit/84e1d355131568529b801d7b30bd5cccac6b3c96) and discards 0.6.0 in the call referenced in my previous comment.

PeterBowman, 2018-06-10:
Please check #1739 against your sample project.

drdanz, 2018-06-25:
@traversaro is this still an issue with latest master?

traversaro, 2018-06-25:
Given that robot-testing and icub-tests seem to be working fine in `devel` (I think @diegoferigo checked this) I think this issue can be closed.

closed 2018-06-25 by traversaro