robotology/yarp #1742: Python bindings broken on master

diegoferigo, 2018-06-12:
**Describe the bug**
Python bindings do not compile.

**To Reproduce**
Steps to reproduce the behavior:

Enable `YARP_COMPILE_BINDINGS` and `CREATE_PYTHON`.

**Expected behavior**

The target `_yarp_python` to compile.

**Configuration (please complete the following information):**
 - OS: Ubuntu 16.04
 - yarp version: d4e1c6d2d8b539b8aef8bcbd49141fa3a83f4cd0
-  compiler: clang++-6.0

**Additional context**

```
$ make _yarp_python
[  1%] Built target YARP_priv_hmac
[ 35%] Built target YARP_OS
[ 37%] Built target YARP_init
[ 41%] Built target YARP_sig
[ 42%] Built target yarpidl_rosmsg
[ 44%] Built target YARP_rosmsg_gen_trajectory_msgs
[ 48%] Built target YARP_math
[ 48%] Built target YARP_rosmsg_gen_tf
[ 50%] Built target YARP_rosmsg_gen_visualization_msgs
[ 51%] Built target YARP_rosmsg_gen_diagnostic_msgs
[ 51%] Built target YARP_rosmsg_gen_native
[ 51%] Built target YARP_rosmsg_gen_tf2_msgs
[ 58%] Built target YARP_rosmsg_gen_geometry_msgs
[ 60%] Built target YARP_rosmsg_gen_actionlib_msgs
[ 62%] Built target YARP_rosmsg_gen_nav_msgs
[ 69%] Built target YARP_rosmsg_gen_sensor_msgs
[ 71%] Built target YARP_rosmsg_gen_shape_msgs
[ 78%] Built target YARP_rosmsg_gen_std_msgs
[ 80%] Built target YARP_rosmsg_gen_stereo_msgs
[ 80%] Built target YARP_rosmsg_gen_all
[ 98%] Built target YARP_dev
[ 98%] Building CXX object bindings/python/CMakeFiles/_yarp_python.dir/__/__/lib/python/yarpPYTHON_wrap.cxx.o
/home/dferigo/iit/sources/yarp/build2/lib/python/yarpPYTHON_wrap.cxx:1489:25: warning: equality comparison with extraneous parentheses [-Wparentheses-equality]
    if (((obj)->ob_type == &PyClass_Type)) {
         ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
/home/dferigo/iit/sources/yarp/build2/lib/python/yarpPYTHON_wrap.cxx:1489:25: note: remove extraneous parentheses around the comparison to silence this warning
    if (((obj)->ob_type == &PyClass_Type)) {
        ~               ^               ~
/home/dferigo/iit/sources/yarp/build2/lib/python/yarpPYTHON_wrap.cxx:1489:25: note: use '=' to turn this equality comparison into an assignment
    if (((obj)->ob_type == &PyClass_Type)) {
                        ^~
                        =
/home/dferigo/iit/sources/yarp/build2/lib/python/yarpPYTHON_wrap.cxx:2291:25: warning: equality comparison with extraneous parentheses [-Wparentheses-equality]
  if (((pyobj)->ob_type == &PyInstance_Type)) {
       ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
/home/dferigo/iit/sources/yarp/build2/lib/python/yarpPYTHON_wrap.cxx:2291:25: note: remove extraneous parentheses around the comparison to silence this warning
  if (((pyobj)->ob_type == &PyInstance_Type)) {
      ~                 ^                  ~
/home/dferigo/iit/sources/yarp/build2/lib/python/yarpPYTHON_wrap.cxx:2291:25: note: use '=' to turn this equality comparison into an assignment
  if (((pyobj)->ob_type == &PyInstance_Type)) {
                        ^~
                        =
/home/dferigo/iit/sources/yarp/build2/lib/python/yarpPYTHON_wrap.cxx:2429:23: warning: explicitly assigning value of variable of type 'int' to itself [-Wself-assign]
                  res = (res);
                  ~~~ ^  ~~~
/home/dferigo/iit/sources/yarp/build2/lib/python/yarpPYTHON_wrap.cxx:2432:23: warning: explicitly assigning value of variable of type 'int' to itself [-Wself-assign]
                  res = (res);
                  ~~~ ^  ~~~
/home/dferigo/iit/sources/yarp/build2/lib/python/yarpPYTHON_wrap.cxx:12388:9: warning: variable 'result' is used uninitialized whenever 'if' condition is true [-Wsometimes-uninitialized]
[... 70 lines cut ...]
/home/dferigo/iit/sources/yarp/build2/lib/python/yarpPYTHON_wrap.cxx:152945:92: error: use of undeclared identifier 'a'
  SWIG_Python_SetConstant(d, "VOCAB_SHARPNESS",SWIG_From_int(static_cast< int >(r*16777216+a*65536+h*256+s)));
                                                                                           ^
/home/dferigo/iit/sources/yarp/build2/lib/python/yarpPYTHON_wrap.cxx:152945:100: error: use of undeclared identifier 'h'
  SWIG_Python_SetConstant(d, "VOCAB_SHARPNESS",SWIG_From_int(static_cast< int >(r*16777216+a*65536+h*256+s)));
                                                                                                   ^
/home/dferigo/iit/sources/yarp/build2/lib/python/yarpPYTHON_wrap.cxx:152945:106: error: use of undeclared identifier 's'
  SWIG_Python_SetConstant(d, "VOCAB_SHARPNESS",SWIG_From_int(static_cast< int >(r*16777216+a*65536+h*256+s)));
                                                                                                         ^
/home/dferigo/iit/sources/yarp/build2/lib/python/yarpPYTHON_wrap.cxx:152946:77: error: use of undeclared identifier 't'
  SWIG_Python_SetConstant(d, "VOCAB_WHITE",SWIG_From_int(static_cast< int >(t*16777216+i*65536+h*256+w)));
                                                                            ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
6 warnings and 20 errors generated.
bindings/python/CMakeFiles/_yarp_python.dir/build.make:71: recipe for target 'bindings/python/CMakeFiles/_yarp_python.dir/__/__/lib/python/yarpPYTHON_wrap.cxx.o' failed
make[3]: *** [bindings/python/CMakeFiles/_yarp_python.dir/__/__/lib/python/yarpPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:5416: recipe for target 'bindings/python/CMakeFiles/_yarp_python.dir/all' failed
make[2]: *** [bindings/python/CMakeFiles/_yarp_python.dir/all] Error 2
CMakeFiles/Makefile2:5428: recipe for target 'bindings/python/CMakeFiles/_yarp_python.dir/rule' failed
make[1]: *** [bindings/python/CMakeFiles/_yarp_python.dir/rule] Error 2
Makefile:1276: recipe for target '_yarp_python' failed
make: *** [_yarp_python] Error 2
```

cc @anqingd

jgvictores, 2018-06-12:
Hi, which SWIG version are you using?

Are you using any SWIG variant, e.g. specific commit/branch for MATLAB bindings?

traversaro, 2018-06-12:
@diegoferigo @anqingd Did you tried on a new build or it was an old build that you updated?

diegoferigo, 2018-06-12:
@traversaro Yes it happened in a clean build folder

@jgvictores Yes, I need more often generating Matlab bindings than python bindings and I keep always installed in my system the https://github.com/jaeandersson/swig fork

traversaro, 2018-06-12:
@diegoferigo @nunoguedelha has patched the swig version that we use to generate matlab bindings with several fixes, we still need to updated the docs but I think we are going to use https://github.com/robotology-dependencies/swig/tree/matlab for the near future, please check with @nunoguedelha for more details.

Nicogene, 2018-06-12:
On my pc it compiles fine, the only difference is that I have an older version of clang.
```
$ clang --version
clang version 3.8.0-2ubuntu4 (tags/RELEASE_380/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

```

diegoferigo, 2018-06-12:
@traversaro With that swig version python bindings compile well! Yes, this should be documented somewhere.

Thank you guys, closing.

closed 2018-06-12 by diegoferigo