robotology/yarp #1746: Static YARP 3.0 is unable to find Eigen3 *twice*

PeterBowman, 2018-06-13:
**Describe the bug**
There is a bug in [*FindEigen3.cmake*](https://github.com/robotology/yarp/blob/74a526fb8f475811cc8af18247b68f2cf3c2f14e/cmake/ycm-0.8.0/3rdparty/FindEigen3.cmake) that spoils successive CMake configure steps on systems with Eigen3 while building static YARP.

**To Reproduce**
* Get latest YARP master (3.0 tag or 74a526f).
* Build as static libs (`cmake .. -DBUILD_SHARED_LIBS=OFF`, compile and install).
* Create a basic downstream project (all you need is the `find_package()` call, though):
  ```cmake
  cmake_minimum_required(VERSION 3.5 FATAL_ERROR)
  project(yarp-test LANGUAGES C)
  find_package(YARP REQUIRED)
  file(WRITE "${CMAKE_BINARY_DIR}/mylib.c" "")
  add_library(mylib mylib.c)
  ```
* Configure once (`cmake ..`) -> success.
* Configure again:
  ```
  CMake Error at /usr/local/lib/cmake/YARP_math/YARP_mathConfig.cmake:27 (find_package):
  Found package configuration file:

    /usr/local/lib/cmake/YARP_eigen/YARP_eigenConfig.cmake

  but it set YARP_eigen_FOUND to FALSE so package "YARP_eigen" is considered
  to be NOT FOUND.  Reason given by package:

  YARP_eigen could not be found because dependency Eigen3 could not be found.
  ```

**Configuration (please complete the following information):**
 - OS: Ubuntu Xenial, CMake 3.5
 - yarp version: 3.0
-  compiler: irrelevant, this is a CMake issue

**Additional context**
I already identified this at https://github.com/robotology/yarp/pull/1731#issuecomment-396094810. The YARP_eigen component issues a `find_dependency(Eigen3)` call that in turn processes the *FindEigen3.cmake* shipped along with YARP's CMake modules. On the first run, a `Eigen3_FOUND` variable is set and correctly interpreted by `find_dependency()`. However, on successive runs this module already sees a `EIGEN3_INCLUDE_DIR` cached variable and sets `EIGEN3_FOUND` (note the all caps, [ref](https://github.com/robotology/yarp/blob/74a526fb8f475811cc8af18247b68f2cf3c2f14e/cmake/ycm-0.8.0/3rdparty/FindEigen3.cmake#L66)). This variable is not recognized by `find_dependency()` anymore and reports missing Eigen3.

It doesn't matter whether YARP was built using it's own internal YCM modules or a YCM installation was found on system. Both situations end up with the same faulty *FindEigen3.cmake* module used by the downstream project (either in `/usr/local/share/yarp/cmake/ycm-0.8.0/3rdparty` or `/usr/local/share/YCM/3rdparty`).

PeterBowman, 2018-06-13:
cc @Nicogene per the referenced comment. Unit tests are failing since [this Travis build](https://travis-ci.org/robotology/yarp/builds/387342747?utm_source=github_status&utm_medium=notification) and a similar error message is shown.

drdanz, 2018-06-25:
@PeterBowman Is this still an issue with latest master?

PeterBowman, 2018-06-25:
On Xenial (Eigen<3.3.1), not anymore since [this line was commented out](https://github.com/robotology/yarp/commit/cd5755b99e21678020178faa5efa3e8696b08bb6#diff-e90993d9a34522763353d1b9bd63da57R30). With Eigen>=3.3.1 (tested: 3.3.90) and CMake 3.9 or later, the test CMake list file I included in the issue description works fine, but the following raises the same error:

```cmake
cmake_minimum_required(VERSION 3.5 FATAL_ERROR)
project(yarp-test LANGUAGES C)
set(EIGEN3_INCLUDE_DIR "/path/to/eigen3/include/dir") # new
find_package(YARP REQUIRED)
file(WRITE "${CMAKE_BINARY_DIR}/mylib.c" "")
add_library(mylib mylib.c)
```

All this boils down to *FindEigen3.cmake* not coping well with the `find_dependency(Eigen3 ...)` call located in *YARP_eigenConfig.cmake*, as described earlier. Compare [this instruction](https://github.com/robotology/yarp/blob/d3fb843e513b637ebb79838e825d3fa60820adb4/cmake/ycm-0.8.1/3rdparty/FindEigen3.cmake#L66) (sets `EIGEN3_FOUND`) and [this check](https://github.com/Kitware/CMake/blob/v3.9.6/Modules/CMakeFindDependencyMacro.cmake#L46-L50) (expects `Eigen3_FOUND`).

To sum up: not an issue with current master **now**, but might inadvertently mess things up again in the future. I'd just add a `set(Eigen3_FOUND ${EIGEN3_FOUND})` there.

PeterBowman, 2018-06-25:
I'd close this issue as invalid since the description and steps to reproduce it have varied notably. Now, it doesn't matter whether YARP is built statically, only affects CMake 3.9+ and modern CMake, may raise an error on the first `cmake ..` run.

@drdanz please consider applying the changes proposed in my previous comment.

drdanz, 2018-07-11:
@PeterBowman I opened a PR on the Eigen3 repository: https://bitbucket.org/eigen/eigen/pull-requests/419/
If that's approved we can easily import it in YCM, but then we should also consider that the latest FindEigen3 supports target, see [this commit](https://bitbucket.org/eigen/eigen/commits/7334ee193567c388e77d67de77631971e87fe61c)

PeterBowman, 2018-07-11:
Thanks! I think there is nothing more to do here, closing.

closed 2018-07-11 by PeterBowman

PeterBowman, 2018-07-12:
Follow-up: https://github.com/robotology/yarp/pull/1800.